mu-editor/mu #1762: Crash Report from a students WIndows machine

KellyPared, 2021-08-20:
This is all I got from the student.  But what happened was when the student first launched Mu, it was unresponsive. He may have clicked multiple times and then 20+ blank windows opened up and he got a crash report. I will try to dig the report out of his MU on Monday.
In the end, the app opened.

**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\alex.bogatov\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\alex.bogatov\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\alex.bogatov\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Users\alex.bogatov\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 767, in create
    self.create_venv()
  File "C:\Users\alex.bogatov\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 798, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\alex.bogatov\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\alex.bogatov\AppData\Local\python\mu\mu_venv-38-20210820-084917
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\alex.bogatov\\AppData\\Local\\python\\mu\\mu_venv-38-20210820-084917\\Scripts\\python.exe'

```

carlosperate, 2021-08-20:
Thank you for the report @KellyPared!
If you could recover the log file that would be really useful too.

Based on your description I think it might be this issue, but seeing to log file would help confirm it: https://github.com/mu-editor/mu/issues/1755

After getting the log, could you try to launch Mu again, ensuring only one instance is running?
Hopefully it should work, but if it doesn't then that would indicate it might be a different problem.
In that case getting a new copy of the log (with the latest crash) would be very useful too.

About he multiple blank windows opening, we've captured that issue here: https://github.com/mu-editor/mu/issues/1457
I'll try to get some free time soon to implement that fix.

KellyPared, 2021-08-20:
We were able to launch it successfully afterwards with only one instance.  I will try to remember to get it on Monday. 
Kelly

carlosperate, 2021-08-20:
Awesome, thanks Kelly!

ntoll, 2021-08-21:
Top bug reporting there @KellyPared (thank you @carlosperate for the triage). :+1:

Hope the weather is good for the school summer holidays in Florida!

KellyPared, 2021-08-21:
Thanks Nick. These 6th graders know how to break things!

Let’s catch up as soon as the first of the year craziness is over!

carlosperate, 2021-09-21:
I assume the log will be hard to retrieve by now.
I'll close this issue as it looks like it might be the "two Mu instances racing each other" problem and that is tracked in https://github.com/mu-editor/mu/issues/1755.

Thanks for the report @KellyPared!

closed 2021-09-21 by carlosperate