mu-editor/mu #1766: Crash Report

jbdblack, 2021-08-24:
[mu.log](https://github.com/mu-editor/mu/files/7035825/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\joryd\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\joryd\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\joryd\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Users\joryd\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 767, in create
    self.create_venv()
  File "C:\Users\joryd\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 798, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\joryd\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\joryd\AppData\Local\python\mu\mu_venv-38-20210823-203423
ModuleNotFoundError: No module named 'virtualenv.seed.embed.via_app_data'

```

carlosperate, 2021-09-29:
Thanks for sending the additional info @jbdblack!

By any chance do you know how many Python installations do you have and do you remember how you installed them? downloaded a installer from the Python website, the Windows app store, or any other method perhaps?

-----

I'll summarise the current findings here.

From your OS PATH we can see these Python related entries:
```
C:\Users\joryd\AppData\Local\Programs\Python\Python39\Scripts\
C:\Users\joryd\AppData\Local\Programs\Python\Python39\
C:\Users\joryd\AppData\Local\Programs\Python\Python38-32\Scripts\
C:\Users\joryd\AppData\Local\Programs\Python\Python38-32\
C:\Users\joryd\AppData\Roaming\Python\Python38\Scripts
```

From Mu's Python `sys.path` we have this entry for a different Python environment:
```
'C:\\Users\\joryd\\AppData\\Roaming\\Python\\Python38\\site-packages'
```

And from Mu's pip we see the following packages are being picked up from the wrong Python environment:

```
Package                Version    Location                                                                                             Installer
---------------------- ---------- ---------------------------------------------------------------------------------------------------- ---------
appdirs                1.4.4      c:\users\joryd\appdata\roaming\python\python38\site-packages                                         pip
certifi                2020.4.5.2 c:\users\joryd\appdata\roaming\python\python38\site-packages                                         pip
distlib                0.3.0      c:\users\joryd\appdata\roaming\python\python38\site-packages
filelock               3.0.12     c:\users\joryd\appdata\roaming\python\python38\site-packages                                         pip
pipenv                 2020.6.2   c:\users\joryd\appdata\roaming\python\python38\site-packages                                         pip
six                    1.15.0     c:\users\joryd\appdata\roaming\python\python38\site-packages                                         pip
virtualenv             20.0.21    c:\users\joryd\appdata\roaming\python\python38\site-packages                                         pip
virtualenv-clone       0.5.4      c:\users\joryd\appdata\roaming\python\python38\site-packages                                         pip
```

The issue in this case is that Mu is picking up `virtualenv` from the system Python 3.8 site-packages, which is a newer version and has moved things around, so Mu cannot find `virtualenv.seed.embed.via_app_data` and it throws that exception.

carlosperate, 2021-12-15:
@jbdblack  We think we might have found a solution to this problem. Would you be able to try the installer at the bottom of this page, in the "Artifacts" section to see if it resolves the problem you are encountering?
https://github.com/mu-editor/mu/actions/runs/1574840282

To ensure things go smoothly it'd be highly recommended to first uninstall the version of Mu in your PC before installing this version.

If you could try this installer it would be very very useful to us to confirm the issue has been fixed.