mu-editor/mu #1770: Crash Report

abel503, 2021-08-25:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: es_MX
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 365, in run
    editor.restore_session(sys.argv[1:])
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 888, in restore_session
    self.direct_load(old_path)
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 1145, in direct_load
    self._load(path)
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 1017, in _load
    text, newline = read_and_decode(path)
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 318, in read_and_decode
    text = btext.decode(encoding)
LookupError: unknown encoding: 2

```

carlosperate, 2021-08-25:
Thanks for the report @abel503!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\PC\AppData\Local\python\mu\Logs\mu.log`

What action were you taking in Mu when it crashed? Were you trying to run a Python script in the "Python 3" mode?
If so, do you have a copy of the file you can add to this issue?

carlosperate, 2021-09-22:
Hi @abel503, just noticed we also have https://github.com/mu-editor/mu/issues/1780 open, so let's continue the debugging over there 👍

closed 2021-09-22 by carlosperate