mu-editor/mu #1777: Crash Report

PanCzaro, 2021-08-28:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Linux 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l

```
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mu/modes/python3.py", line 109, in start_kernel
    self.repl_kernel_manager.start_kernel()
  File "/home/pi/.local/lib/python3.7/site-packages/mu/modes/python3.py", line 42, in start_kernel
    kernel_cmd, kw = self.pre_start_kernel(**kw)
AttributeError: 'MuKernelManager' object has no attribute 'pre_start_kernel'

```

carlosperate, 2021-08-30:
Hi @PanCzaro,

Thank you for the report. I believe this issue might be this: https://github.com/mu-editor/mu/pull/1517
We are looking into a fix and the next Mu release should hopefully work out-of-the-box.

In the meantime, the easiest way to fix this would be to update the `jupyter_client` package to a newer version.
However, since the reason you have an old version is because some other application in your computer is using it, there is a risk that updating it could cause issues in that other application. Updating your system `jupyter_client` is the easiest solution, but it does have that risk.

You can install Mu in a virtual environment, this isolates the Mu installation, so that its dependencies are not combined with the system packages. 
If you are familiar with virtual environments you can create a new one for Mu, install Mu using pip and it should pick up the latest version of  `jupyter_client`.

Alternatively I like to use the pipx tool, which simplifies all this process: https://pipxproject.github.io/pipx/
If you install it you can then use pipx to automatically install Python applications (like Mu) in their own isolated virtual environment, and they can be run without manually activating the environment:
- Install pipx
    - Update pip: `python3 -m pip install --upgrade pip` 
    - Run: `python3 -m pip install --user pipx`
    - Run: `python3 -m pipx ensurepath`
- After installation you might need to close all your terminals and open a new one (or maybe even reboot, the last command should print out information)
- Now that is pipx is installed and configured you can use it to install Mu: `pipx install mu-editor==1.1.0b5`
- And you can run mu: `mu-editor`

warybyte, 2021-09-23:
I was having issues with my Python3 virtual env loading after installing mu-editor. This tip on installing pipx then reinstalling mu-editor worked for me.

Linux server 5.4.0-84-generic #94~18.04.1-Ubuntu SMP Thu Aug 26 23:17:46 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux