mu-editor/mu #1780: Crash Report

abel503, 2021-08-30:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: es_MX
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 365, in run
    editor.restore_session(sys.argv[1:])
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 888, in restore_session
    self.direct_load(old_path)
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 1145, in direct_load
    self._load(path)
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 1017, in _load
    text, newline = read_and_decode(path)
  File "C:\Users\PC\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 318, in read_and_decode
    text = btext.decode(encoding)
LookupError: unknown encoding: 2

```

carlosperate, 2021-08-30:
Thanks for the report @abel503!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\PC\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-09-22:
@abel503 the Mu log would be really useful for us to help solve this problem, could you add it to this GitHub issue?
File should be located in `C:\Users\PC\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-09-23:
Hi @abel503,

Could you try the installer "mu-editor-Windows-0a954b6da7f4ccc4c8f3519bdfc3f2e49c0589d8" from this page? https://github.com/mu-editor/mu/actions/runs/1266857664

I think I might have fixed this issue in https://github.com/mu-editor/mu/pull/1817 and it would be really useful if you could test it with that installer.

Thanks!

carlosperate, 2021-10-04:
Thank you for the report @abel503, this should be fixed in the next beta release of Mu. 👍

closed 2021-10-04 by carlosperate

abel503, 2021-10-04:
oh thanks for the help, i hadn't checked my email sorry, have a happy day
and use MU !!!