mu-editor/mu #1781: Crash Report

elias40000, 2021-08-30:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: es_SV
Platform: Windows 10 10.0.16299 AMD64

```
"C:\Users\usuario\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\usuario\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\usuario\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\usuario\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\usuario\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 365, in run
    editor.restore_session(sys.argv[1:])
  File "C:\Users\usuario\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 888, in restore_session
    self.direct_load(old_path)
  File "C:\Users\usuario\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py

carlosperate, 2021-08-30:
Thanks for the report @elias40000!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\usuario\AppData\Local\python\mu\Logs\mu.log`

elias40000, 2021-09-10:
2021-09-09 08: 55: 20,828 - mu.settings: 169 (save) DEBUG: Guardando
en C: \ Users \ usuario \ AppData \ Local \ python \ mu \ session.json
2021-09-09 08: 55: 20,831 - mu.settings: 169 (save) DEBUG: Guardando
en C: \ Users \ usuario \ AppData \ Local \ python \ mu \ venv.json
2021-09-09 17: 44: 47,724 - root: 271 (ejecutar) INFO:
-----------------
Inicio de Mu 1.1.0.beta.5
2021-09-09 17: 44: 47,725 - root: 272 (ejecutar) INFO: uname_result
(system = 'Windows', node = 'LAPTOP-QC8FGO8S', release = '10 ',
version = '10 .0.16299', machine = 'AMD64', procesador = 'Intel64
Familia 6 Modelo 78 Paso 3, GenuineIntel')
2021-09-09 17: 44: 47,725 - root: 273 (ejecutar) INFO: Plataforma:
Windows-10-10.0.16299-SP0
2021-09-09 17: 44: 47,725 - root: 274 (ejecutar) INFO: Ruta de Python:
['C: \\ Users \\ usuario \\ AppData \\ Local \\ Programs \\ Mu
Editor', 'C: \\ Usuarios \\ usuario \\ AppData \\ Local \\ Programas
\\ Mu Editor \\ Python \\ python38.zip ',' C: \\ Users \\ usuario \\
AppData \\ Local \\ Programas \\ Mu Editor \\ Python \\ DLLs ',' C: \\
Users \\ usuario \\ AppData \\ Local \\ Programs \\ Mu Editor \\
Python \\ lib ',' C: \\ Users \\ usuario \\ AppData \\ Local \\
Programas \\ Mu Editor \\ Python ',' C: \\ Users \\ usuario \\ AppData
\\ Local \\ Programas \\ Mu Editor \\ Python \\ lib \\ site-packages '
, 'C: \\ Usuarios \\ usuario \\ AppData \\ Local \\ Programas \\ Mu
Editor \\ Python \\ lib \\ site-packages \\
setuptools-49.6.0-py3.8.egg', ' C: \\ Usuarios \\ usuario \\ AppData
\\ Local \\ Programas \\ Mu Editor \\ Python \\ lib \\ site-packages
\\ pip-20.2.2-py3.8.egg ',' C:\\ Usuarios \\ usuario \\ AppData \\
Local \\ Programas \\ Mu Editor \\ Python \\ lib \\ site-packages \\
win32 ',' C: \\ Users \\ usuario \\ AppData \\ Local \\ Programas \\
Mu Editor \\ Python \\ lib \\ site-packages \\ win32 \\ lib ',' C: \\
Users \\ usuario \\ AppData \\ Local \\ Programas \\ Mu Editor \\
Python \\ lib \\ paquetes de sitio \\ Pythonwin ',' C: \\ Usuarios \\
usuario \\ AppData \\ Local \\ Programas \\ Mu Editor \\ Python \\ lib
\\ paquetes de sitio \\ IPython \\ extensiones ']
2021-09-09 17: 44: 47,725 - root: 274 (ejecutar) INFO: Ruta de Python:
['C: \\ Users \\ usuario \\ AppData \\ Local \\ Programs \\ Mu
Editor', 'C: \\ Usuarios \\ usuario \\ AppData \\ Local \\ Programas
\\ Mu Editor \\ Python \\ python38.zip ',' C: \\ Users \\ usuario \\
AppData \\ Local \\ Programas \\ Mu Editor \\ Python \\ DLLs ',' C: \\
Users \\ usuario \\ AppData \\ Local \\ Programs \\ Mu Editor \\
Python \\ lib ',' C: \\ Users \\ usuario \\ AppData \\ Local \\
Programas \\ Mu Editor \\ Python ',' C: \\ Users \\ usuario \\ AppData
\\ Local \\ Programas \\ Mu Editor \\ Python \\ lib \\ site-packages '
, 'C: \\ Usuarios \\ usuario \\ AppData \\ Local \\ Programas \\ Mu
Editor \\ Python \\ lib \\ site-packages \\
setuptools-49.6.0-py3.8.egg', ' C: \\ Usuarios \\ usuario \\ AppData
\\ Local \\ Programas \\ Mu Editor \\ Python \\ lib \\ site-packages
\\ pip-20.2.2-py3.8.egg ',' C:\\ Usuarios \\ usuario \\ AppData \\
Local \\ Programas \\ Mu Editor \\ Python \\ lib \\ site-packages \\
win32 ',' C: \\ Users \\ usuario \\ AppData \\ Local \\ Programas \\
Mu Editor \\ Python \\ lib \\ site-packages \\ win32 \\ lib ',' C: \\
Users \\ usuario \\ AppData \\ Local \\ Programas \\ Mu Editor \\
Python \\ lib \\ paquetes de sitio \\ Pythonwin ',' C: \\ Usuarios \\
usuario \\ AppData \\ Local \\ Programas \\ Mu Editor \\ Python \\ lib
\\ paquetes de sitio \\ IPython \\ extensiones ']
2021-09-09 17: 44: 47,726 - root: 275 (ejecutar) INFO: Código de idioma: es_SV
2021-09-09 17: 44: 47,726 - mu.settings: 220 (cargar) ADVERTENCIA: No
se encontró ningún archivo de configuración en C: \ Users \ usuario \
AppData \ Local \ python \ mu \ settings.json; salto a la comba
2021-09-09 17: 44: 47,726 - mu.settings: 220 (cargar) ADVERTENCIA: No
se encontró ningún archivo de configuración en C: \ Users \ usuario \
AppData \ Local \ python \ mu \ settings.json; salto a la comba
2021-09-09 17: 44: 48,300 - mu.virtual_environment: 585
(asegurar_y_crear) INFO: Controlador de registro agregado.
2021-09-09 17: 44: 48,314 - mu.virtual_environment: 585
(asegurar_y_crear) INFO: Controlador de registro agregado.
2021-09-09 17: 44: 48,763 - mu.virtual_environment: 596
(asegurar_y_crear) DEBUG: Comprobando el entorno virtual; intento # 1.
2021-09-09 17: 44: 48,764 - mu.virtual_environment: 596
(asegurar_y_crear) DEBUG: Comprobando el entorno virtual; intento # 1.
[... 331 lines cut ...]
restore_session
    self.direct_load (ruta_antigua)
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programas \ Mu
Editor \ Python \ lib \ site-packages \ mu \ logic.py", línea 1145, en
direct_load
    self._load (ruta)
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programs \ Mu
Editor \ Python \ lib \ site-packages \ mu \ logic.py", línea 1017, en
_load
    texto, nueva línea = read_and_decode (ruta)
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programs \ Mu
Editor \ Python \ lib \ site-packages \ mu \ logic.py", línea 318, en
read_and_decode
    text = btext.decode (codificación)
LookupError: codificación desconocida: uft-8
2021-09-09 17: 56: 00,132 - mu.settings: 169 (save) DEBUG: Guardando
en C: \ Users \ usuario \ AppData \ Local \ python \ mu \ session.json
2021-09-09 17: 56: 00,133 - mu.settings: 169 (save) DEBUG: Guardando
en C: \ Users \ usuario \ AppData \ Local \ python \ mu \ venv.json

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

carlosperate, 2021-09-22:
This is the interesting part:

```
2021-09-09 17: 44: 54,306 - mu.logic: 1523 (change_mode) INFO: Directorio del área de trabajo: C: \ Users \ usuario \ mu_code
2021-09-09 17: 44: 54,306 - mu.logic: 867 (restore_session) DEBUG: <SessionSettings de C: \ Users \ usuario \ AppData \ Local \ python \ mu \ session.json>
2021-09-09 17: 44: 54,317 - mu.logic: 978 (_load) INFO: Cargando script desde: C: \ Users \ usuario \ AppData \ Local \ Programs \ Mu Editor \ .py
2021-09-09 17: 44: 54,334 - mu.logic: 308 (read_and_decode) DEBUG: Codificación detectada uft-8
2021-09-09 17: 44: 54,334 - mu.logic: 316 (read_and_decode) DEBUG: Intentando decodificar con uft-8
2021-09-09 17: 44: 54,335 - root: 173 (excepto gancho) ERROR: Error irrecuperable
Rastreo (llamadas recientes más última):
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programs \ Mu Editor \ Python \ lib \ runpy.py", línea 194, en _run_module_as_main
    return _run_code (código, main_globals, None,
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programs \ Mu Editor \ Python \ lib \ runpy.py", línea 87, en _run_code
    exec (código, run_globals)
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programas \ Mu Editor \ Python \ lib \ site-packages \ mu \ __ main__.py", línea 9, en <módulo>
    principal()
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programas \ Mu Editor \ Python \ lib \ site-packages \ mu \ __ main__.py", línea 5, en main
    correr()
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programas \ Mu Editor \ Python \ lib \ site-packages \ mu \ app.py", línea 365, en ejecución
    editor.restore_session (sys.argv [1:])
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programs \ Mu Editor \ Python \ lib \ site-packages \ mu \ logic.py", línea 888, en restore_session
    self.direct_load (ruta_antigua)
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programas \ Mu Editor \ Python \ lib \ site-packages \ mu \ logic.py", línea 1145, en direct_load
    self._load (ruta)
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programs \ Mu Editor \ Python \ lib \ site-packages \ mu \ logic.py", línea 1017, en _load
    texto, nueva línea = read_and_decode (ruta)
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programs \ Mu Editor \ Python \ lib \ site-packages \ mu \ logic.py", línea 318, en read_and_decode
    text = btext.decode (codificación)
LookupError: codificación desconocida: uft-8
```

```
2021-09-09 17: 55: 59,278 - mu.logic: 978 (_load) INFO: Cargando script desde: C: \ Users \ usuario \ AppData \ Local \ Programs \ Mu Editor \ .py
2021-09-09 17: 55: 59,279 - mu.logic: 308 (read_and_decode) DEBUG: Codificación detectada uft-8
2021-09-09 17: 55: 59,279 - mu.logic: 316 (read_and_decode) DEBUG: Intentando decodificar con uft-8
2021-09-09 17: 55: 59,280 - root: 173 (excepto gancho) ERROR: Error irrecuperable
Rastreo (llamadas recientes más última):
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programs \ Mu Editor \ Python \ lib \ runpy.py", línea 194, en _run_module_as_main
    return _run_code (código, main_globals, None,
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programs \ Mu Editor \ Python \ lib \ runpy.py", línea 87, en _run_code
    exec (código, run_globals)
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programas \ Mu Editor \ Python \ lib \ site-packages \ mu \ __ main__.py", línea 9, en <módulo>
    principal()
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programas \ Mu Editor \ Python \ lib \ site-packages \ mu \ __ main__.py", línea 5, en main
    correr()
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programas \ Mu Editor \ Python \ lib \ site-packages \ mu \ app.py", línea 365, en ejecución
    editor.restore_session (sys.argv [1:])
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programs \ Mu Editor \ Python \ lib \ site-packages \ mu \ logic.py", línea 888, en restore_session
    self.direct_load (ruta_antigua)
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programas \ Mu Editor \ Python \ lib \ site-packages \ mu \ logic.py", línea 1145, en direct_load
    self._load (ruta)
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programs \ Mu Editor \ Python \ lib \ site-packages \ mu \ logic.py", línea 1017, en _load
    texto, nueva línea = read_and_decode (ruta)
  Archivo "C: \ Users \ usuario \ AppData \ Local \ Programs \ Mu Editor \ Python \ lib \ site-packages \ mu \ logic.py", línea 318, en read_and_decode
    text = btext.decode (codificación)
LookupError: codificación desconocida: uft-8
```

@elias40000 I believe your email client might be translating some of the text from the log, which makes it difficult to interpret. Would you be able to drag & drop the log file into the GitHub issue?
Via the GitHub website you can upload the file itself and it'll be very helpful for us to analyse it.

Gracias!

carlosperate, 2021-09-23:
Hi @elias40000,

Could you try the installer "mu-editor-Windows-0a954b6da7f4ccc4c8f3519bdfc3f2e49c0589d8" from this page? https://github.com/mu-editor/mu/actions/runs/1266857664

I think I might have fixed this issue in https://github.com/mu-editor/mu/pull/1817 and it would be really useful if you could test it with that installer.

Thanks!

carlosperate, 2021-10-04:
Thank you for the report @elias40000, this should be fixed in the next beta release of Mu. 👍

closed 2021-10-04 by carlosperate