mu-editor/mu #1783: Crash Report

socalledsound, 2021-08-30:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Darwin 16.7.0 Darwin Kernel Version 16.7.0: Thu Jun 15 17:36:27 PDT 2017; root:xnu-3789.70.16~2/RELEASE_X86_64 x86_64

```
ate
    self.install_baseline_packages()
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 880, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: debugpy-1.3.0-cp38-cp38-macosx_10_14_x86_64.whl is not a supported wheel on this platform.

```

socalledsound, 2021-08-30:
mu editor crashing on boot on mac 10.12.6

socalledsound, 2021-08-30:
`2021-08-30 11:18:40,208 - root:271(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.5
2021-08-30 11:18:40,208 - root:272(run) INFO: uname_result(system='Darwin', node='douglass-imac.lan', release='16.7.0', version='Darwin Kernel Version 16.7.0: Thu Jun 15 17:36:27 PDT 2017; root:xnu-3789.70.16~2/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-08-30 11:18:40,213 - root:273(run) INFO: Platform: macOS-10.12.6-x86_64-i386-64bit
2021-08-30 11:18:40,213 - root:274(run) INFO: Python path: ['/', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python38.zip', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/lib-dynload', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/setuptools-49.6.0-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/pip-20.2.2-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/IPython/extensions']
2021-08-30 11:18:40,213 - root:275(run) INFO: Language code: en_US
2021-08-30 11:18:40,214 - mu.settings:220(load) WARNING: No settings file found at /Users/df/Library/Application Support/mu/settings.json; skipping
2021-08-30 11:18:40,214 - mu.settings:220(load) WARNING: No settings file found at /Users/df/Library/Application Support/mu/session.json; skipping
2021-08-30 11:18:40,514 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2021-08-30 11:18:40,515 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-08-30 11:18:40,515 - mu.virtual_environment:777(create_venv) INFO: Creating virtualenv: /Users/df/Library/Application Support/mu/mu_venv-38-20210830-111839
2021-08-30 11:18:40,515 - mu.virtual_environment:778(create_venv) INFO: Virtualenv name: mu_venv-38-20210830-111839
2021-08-30 11:18:40,515 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/Applications/Mu Editor.app/Contents/MacOS/../Resources/Python/bin/python3.8', '-m', 'virtualenv', '-p', '/Applications/Mu Editor.app/Contents/MacOS/../Resources/Python/bin/python3.8', '-q', '/Users/df/Library/Application Support/mu/mu_venv-38-20210830-111839') with kwargs {'env': {'SHELL': '/bin/bash', 'TMPDIR': '/var/folders/3p/c_1fnlpn5c34rxn8c37444b80000gn/T/', 'Apple_PubSub_Socket_Render': '/private/tmp/com.apple.launchd.Bb3g1zWJvf/Render', 'USER': 'df', 'COMMAND_MODE': 'unix2003', 'TCL_LIBRARY': '/Applications/Mu Editor.app/Contents/MacOS/../Resources/Python/lib/tcl8.6', 'SSH_AUTH_SOCK': '/private/tmp/com.apple.launchd.jHeayPIPH2/Listeners', '__CF_USER_TEXT_ENCODING': '0x1F5:0x0:0x0', 'PATH': '/usr/bin:/bin:/usr/sbin:/sbin', 'PWD': '/', 'XPC_FLAGS': '0x0', 'XPC_SERVICE_NAME': 'com.apple.xpc.launchd.oneshot.0x10000010.Mu Editor', 'SHLVL': '0', 'HOME': '/Users/df', 'LOGNAME': 'df', 'DISPLAY': '/private/tmp/com.apple.launchd.51bi2Ec4Ud/org.macosforge.xquartz:0', 'LC_CTYPE': 'UTF-8', 'QT_API': 'pyqt5', 'QT_AUTO_SCREEN_SCALE_FACTOR': '1', 'QT_MAC_WANTS_LAYER': '1'}}
2021-08-30 11:18:41,141 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2021-08-30 11:18:41,141 - mu.virtual_environment:792(create_venv) INFO: Created virtual environment using /Applications/Mu Editor.app/Contents/MacOS/../Resources/Python/bin/python3.8 at /Users/df/Library/Application Support/mu/mu_venv-38-20210830-111839
2021-08-30 11:18:41,146 - mu.virtual_environment:868(install_baseline_packages) INFO: Installing baseline packages.
2021-08-30 11:18:41,150 - mu.virtual_environment:875(install_baseline_packages) INFO: Expecting zipped wheels at /Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/wheels/1.1.0.beta.5.zip
2021-08-30 11:18:41,321 - mu.virtual_environment:850(install_from_zipped_wheels) INFO: About to install from wheel: appnope-0.1.2-py2.py3-none-any.whl
2021-08-30 11:18:41,325 - mu.virtual_environment:114(run_blocking) INFO: About to run blocking /Users/df/Library/Application Support/mu/mu_venv-38-20210830-111839/bin/pip with args ['install', '--disable-pip-version-check', '--no-deps', '--no-index', '/var/folders/3p/c_1fnlpn5c34rxn8c37444b80000gn/T/tmpb06o4tr9/appnope-0.1.2-py2.py3-none-any.whl'] and envvars {}
2021-08-30 11:18:42,142 - mu.virtual_environment:164(wait) DEBUG: Finished: True; exitStatus 0; exitCode 0
2021-08-30 11:18:42,142 - mu.virtual_environment:249(run) DEBUG: Process output: Processing /var/folders/3p/c_1fnlpn5c34rxn8c37444b80000gn/T/tmpb06o4tr9/appnope-0.1.2-py2.py3-none-any.whl
Installing collected packages: appnope
Successfully installed appnope-0.1.2
2021-08-30 11:18:42,143 - mu.virtual_environment:850(install_from_zipped_wheels) INFO: About to install from wheel: backcall-0.2.0-py2.py3-none-any.whl
2021-08-30 11:18:42,143 - mu.virtual_environment:114(run_blocking) INFO: About to run blocking /Users/df/Library/Application Support/mu/mu_venv-38-20210830-111839/bin/pip with args ['install', '--disable-pip-version-check', '--no-deps', '--no-index', '/var/folders/3p/c_1fnlpn5c34rxn8c37444b80000gn/T/tmpb06o4tr9/backcall-0.2.0-py2.py3-none-any.whl'] and envvars {}
2021-08-30 11:18:42,469 - mu.virtual_environment:164(wait) DEBUG: Finished: True; exitStatus 0; exitCode 0
[... cut ...]
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: debugpy-1.3.0-cp38-cp38-macosx_10_14_x86_64.whl is not a supported wheel on this platform.
2021-08-30 12:14:54,155 - mu.settings:169(save) DEBUG: Saving to /Users/df/Library/Application Support/mu/session.json
2021-08-30 12:14:54,156 - mu.settings:169(save) DEBUG: Saving to /Users/df/Library/Application Support/mu/venv.json`

carlosperate, 2021-08-30:
Hi @socalledsound, thanks for the report!

Are you in macOS 10.12 Sierra? If so, this is a known issue with one of the dependencies that will be fixed in the next release.
In the meantime, as a workaround, you can try the installer from https://github.com/mu-editor/mu/issues/1683#issuecomment-877058288.

I'll close this issue as we are tracking the problem in https://github.com/mu-editor/mu/issues/1683, but thanks again for the report, it is very much appreciated!

closed 2021-08-30 by carlosperate