mu-editor/mu #1784: Crash Report

Andyfahee, 2021-08-31:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: nb_NO
Platform: Windows 10 10.0.19041 AMD64

```
"C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 768, in create
    self.install_baseline_packages()
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 880, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 177, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process did not terminate normally:

```

carlosperate, 2021-08-31:
Thanks for the report @Andyfahee!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `%LOCALAPPDATA%\python\mu\Logs\`

carlosperate, 2021-09-29:
Hi @Andyfahee, are you still experiencing this issue? Would you be able to send us the logs located in `%LOCALAPPDATA%\python\mu\Logs\`?

carlosperate, 2021-12-17:
Hi @Andyfahee,

Some people have reported a similar issue caused by a running antivirus. Do you have an antivirus installed, and if you do, would you be able to disable it and try again?

If that works it'd be good to know what antivirus software it is so that we can try to replicate and see if we can resolve the issue without having to disable it.

carlosperate, 2022-05-31:
Closing due to lack of activity.

closed 2022-05-31 by carlosperate