mu-editor/mu #1794: Crash Report

JanJansen47, 2021-09-09:
[errorlog.txt](https://github.com/mu-editor/mu/files/7138023/errorlog.txt)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_NL
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Jan_J\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 762, in on_ls_fail
    self.disable()
  File "C:\Users\Jan_J\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\interface\panes.py", line 701, in disable
    self.microbit_fs.setDisabled(True)
RuntimeError: wrapped C/C++ object of type MicroPythonDeviceFileList has been deleted

```

JanJansen47, 2021-09-09:
This concerns an esp8266 put the micropyton bin first with the esptool and ubuntu and thereafter with windows and the MU tool. 
Mu works fine with esp32 ( micropyton) and with esp32S2 circuitpython. esp8266 REPL produces garbage . On Ubuntu the result (Thonny) is oke.

carlosperate, 2021-09-27:
Thank you very much for the crash report @JanJansen47!

I believe this is the same issue from https://github.com/mu-editor/mu/issues/1428, so to try to keep things tidy I will merge this report there to track the progress in a single place.

closed 2021-09-27 by carlosperate