mu-editor/mu #1797: Mu Editor won't start after downloading. Error: No Qt platform plugin could be initialized.

Capekk, 2021-09-10:
### What were you trying to do?

Hi, this was my first attempt of coding in circuitpython, but it didn't go as planned... I wanted to code in circuitpython in Mu, but I can't open the app.

### What steps did you take to trigger the issue?

Downloaded the install file. Installed Mu. Tried to open it.

### What did you expect to happen?

I expected it to open 

### What actually happened?

Mu didn't open.

The error message said:
_This application failed to start because no Qt platform plugin could be initialized._

[image]

### Operating System Version

Windows 10 Home

### Mu Version

1.1.0-beta-5

### Other Info

As I said, I don't understand what happend or how to fix it.

### Editor Log

2021-09-10 16:29:35,601 - root:271(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.5
2021-09-10 16:29:35,601 - root:272(run) INFO: uname_result(system='Windows', node='Number-Cruncher', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 158 Stepping 10, GenuineIntel')
2021-09-10 16:29:35,601 - root:273(run) INFO: Platform: Windows-10-10.0.19041-SP0
2021-09-10 16:29:35,601 - root:274(run) INFO: Python path: ['C:\\Users\\Viktor Javůrek\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\Viktor Javůrek\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\Viktor Javůrek\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\Viktor Javůrek\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\Viktor Javůrek\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\Viktor Javůrek\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\Viktor Javůrek\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\Viktor Javůrek\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\Viktor Javůrek\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\Viktor Javůrek\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\Viktor Javůrek\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\Viktor Javůrek\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-09-10 16:29:35,601 - root:275(run) INFO: Language code: en_US
2021-09-10 16:29:35,602 - mu.settings:220(load) WARNING: No settings file found at C:\Users\VIKTOR~1\AppData\Local\python\mu\settings.json; skipping
2021-09-10 16:29:35,602 - mu.settings:220(load) WARNING: No settings file found at C:\Users\VIKTOR~1\AppData\Local\python\mu\session.json; skipping
2021-09-10 16:40:52,079 - root:271(run) INFO:

Plshaa, 2021-09-22:
I have the same problem and i don't know what to do too.

Capekk, 2021-09-25:
> I have the same problem and i don't know what to do too.

I downloaded the program on another device (my old notebook) and copied it to my PC.
Now I can access the folder I copied it to and start the program from there(or create a shortcut to my desktop). I don't know how I can start "copy-paste" Mu, but most likely the files you and I download get corrupt in some way along downloading.

Plshaa, 2021-09-25:
Helped, thank you. : )