mu-editor/mu #1801: Crash Report

fratquintero, 2021-09-14:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Darwin 17.7.0 Darwin Kernel Version 17.7.0: Fri Oct 30 13:34:27 PDT 2020; root:xnu-4570.71.82.8~1/RELEASE_X86_64 x86_64

```
ate
    self.install_baseline_packages()
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 880, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: debugpy-1.3.0-cp38-cp38-macosx_10_14_x86_64.whl is not a supported wheel on this platform.

```

carlosperate, 2021-09-20:
Hi @fratquintero, thanks for the report!

Are you in macOS 10.13 High Sierra? If so, this is a known issue with one of the dependencies that will be fixed in the next release.
In the meantime, as a workaround, you can try the installer from https://github.com/mu-editor/mu/issues/1683#issuecomment-877058288.

I'll close this issue as we are tracking the problem in https://github.com/mu-editor/mu/issues/1683, but thanks again for the report, it is very much appreciated!

closed 2021-09-20 by carlosperate

fratquintero, 2021-09-20:
> In the meantime, as a workaround, you can try the installer from #1683 (comment).

Hi Carlos,

Indeed, I am in macOS 10.13

I am going to download the installer from #1683 (comment) and see how it goes

Best,

Frat Quintero
304 569-7649
Sent with a Spark