mu-editor/mu #1806: Crash Report

rivandemo, 2021-09-17:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: fr_BE
Platform: Darwin 16.7.0 Darwin Kernel Version 16.7.0: Sun Jun  2 20:26:31 PDT 2019; root:xnu-3789.73.50~1/RELEASE_X86_64 x86_64

```
ate
    self.install_baseline_packages()
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 880, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: debugpy-1.3.0-cp38-cp38-macosx_10_14_x86_64.whl is not a supported wheel on this platform.

```

ajs256, 2021-09-18:
Looks like Mu is assuming anything using the Darwin kernel is macOS.

rivandemo, 2021-09-19:
Might be my OS version: I'm running Sierra

carlosperate, 2021-09-20:
Hi @rivandemo, thanks for the report!

If you are in macOS 10.12 Sierra this is a known issue with one of the dependencies that will be fixed in the next release.
In the meantime, as a workaround, you can try the installer from https://github.com/mu-editor/mu/issues/1683#issuecomment-877058288.

I'll close this issue as we are tracking the problem in https://github.com/mu-editor/mu/issues/1683, but thanks again for the report, it is very much appreciated!

closed 2021-09-20 by carlosperate