mu-editor/mu #1816: Crash Report: ModuleNotFoundError: No module named 'ipython_genutils'

zp5r5z, 2021-09-23:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: pt_BR
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\user\Desktop\Paulo\mu\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\user\Desktop\Paulo\mu\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\user\Desktop\Paulo\mu\mu\virtual_environment.py", line 627, in ensure_and_create
    self.ensure()
  File "C:\Users\user\Desktop\Paulo\mu\mu\virtual_environment.py", line 667, in ensure
    self.ensure_key_modules()
  File "C:\Users\user\Desktop\Paulo\mu\mu\virtual_environment.py", line 746, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: ipykernel
STDERR: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\user\AppData\Local\python\mu\mu_venv-38-20210923-111613\lib\site-packages\ipykernel\__init__.py", line 2, in <module>
    from .connect import *
  File "C:\Users\user\AppData\Local\python\mu\mu_venv-38-20210923-111613\lib\site-packages\ipykernel\connect.py", line 13, in <module>
    from ipython_genutils.path import filefind
ModuleNotFoundError: No module named 'ipython_genutils'

```
Used python3.8 with virtualenv activated, I did run the following commands:
```
pip install -e ".[dev]"
python run.py
```

carlosperate, 2021-09-24:
Thanks for the report @zp5r5z!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\user\AppData\Local\python\mu\Logs\mu.log`

Could you also run these on a command prompt inside the virtual env? It will help us too:

```
python -c "import sys; print(sys.path)"
```

```
python -m pip list --verbose
```

carlosperate, 2021-10-05:
Hi @zp5r5z, the fix for this is in PR https://github.com/mu-editor/mu/pull/1844 and should be working in the incoming next beta release.

Thanks for the report!