mu-editor/mu #1820: Crash Report

dkjrivero, 2021-09-25:
**REMEMBER TO ATTACH YOUR LOG FILE**

[mu.log](https://github.com/mu-editor/mu/files/7228953/mu.log)

Version: 1.1.0.beta.5
Locale: en_US
Platform: Darwin 20.2.0 Darwin Kernel Version 20.2.0: Wed Dec  2 20:39:59 PST 2020; root:xnu-7195.60.75~1/RELEASE_X86_64 x86_64

```
entCreateError: Unable to create a virtual environment using /Applications/Mu Editor.app/Contents/MacOS/../Resources/Python/bin/python3.8 at /Users/dominiquer/Library/Application Support/mu/mu_venv-38-20210924-181437
RuntimeError: failed to build image pip because:
Traceback (most recent call last):
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/virtualenv/seed/embed/via_app_data/via_app_data.py", line 57, in _install
    installer.install(creator.interpreter.version_info)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/virtualenv/seed/embed/via_app_data/pip_install/base.py", line 46, in install
    for name, module in self._console_scripts.items():
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/virtualenv/seed/embed/via_app_data/pip_install/base.py", line 131, in _console_scripts
    entry_points = self._dist_info / "entry_points.txt"
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/virtualenv/seed/embed/via_app_data/pip_install/base.py", line 118, in _dist_info
    raise RuntimeError(msg)  # pragma: no cover
RuntimeError: no .dist-info at /Users/dominiquer/Library/Application Support/virtualenv/wheel/3.8/image/1/CopyPipInstall/pip-21.2.4-py3-none-any, has pip

```

carlosperate, 2021-09-27:
Thanks for the report @dkjrivero!

Could you run the following commands in a terminal and paste the result here to help us debug this issue?

```
"/Applications/Mu Editor.app/Contents/MacOS/../Resources/Python/bin/python3.8" -c "import sys; print(sys.path)"
```

```
"/Applications/Mu Editor.app/Contents/MacOS/../Resources/Python/bin/python3.8" -m pip list --verbose
```

carlosperate, 2021-09-28:
Thanks @dkjrivero, the entries in the Python PATH are alright, and the output of pip verbose also looks good.

This looks similar to this problem: https://github.com/pypa/virtualenv/issues/1971
However, you are on macOS, so it's not related to the path being too long.

Could you try this command first, and send us the output?

```
ls -la "/Users/dominiquer/Library/Application Support/virtualenv/wheel/3.8/image/1/CopyPipInstall/pip-21.2.4-py3-none-any"
```

The output of the `ls` command above should show a `pip-21.2.4.dist-info` directory, if it's not there it would be the cause of the error.

Either way, after that try this to reset those cache files:
```
"/Applications/Mu Editor.app/Contents/MacOS/../Resources/Python/bin/python3.8" -m virtualenv --reset-app-data
```

And then try running Mu again to see if that fixed it.

carlosperate, 2021-11-02:
Hi @dkjrivero,
Just wondering if you are able to try the steps from my previous message? It would really help us diagnose the issue and help other users with the same problem.

carlosperate, 2021-12-15:
@dkjrivero We think we might have found a solution to this problem. Would you be able to try the installer at the bottom of this page, in the "Artifacts" section to see if it resolves the problem you are encountering?
https://github.com/mu-editor/mu/actions/runs/1574840282

To ensure things go smoothly it'd be highly recommended to first uninstall the version of Mu in your PC before installing this version.

If you could try this installer it would be very very useful to us to confirm the issue has been fixed.