mu-editor/mu #1823: Crash Report - just few seconds ago on Ubuntu 20.04

abixadamj, 2021-09-25:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: pl_PL
Platform: Linux 5.11.0-36-generic #40~20.04.1-Ubuntu SMP Sat Sep 18 02:14:19 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/adasiek/mu/mu/app.py", line 159, in run
    raise ex
  File "/home/adasiek/mu/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/adasiek/mu/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/adasiek/mu/mu/virtual_environment.py", line 768, in create
    self.install_baseline_packages()
  File "/home/adasiek/mu/mu/virtual_environment.py", line 880, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "/home/adasiek/mu/mu/virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/home/adasiek/mu/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/home/adasiek/mu/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/home/adasiek/mu/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/home/adasiek/mu/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
[mu.log](https://github.com/mu-editor/mu/files/7229447/mu.log)

```

abixadamj, 2021-09-25:
Strange, because I can instaall it on system...

```
adasiek@adasiek-Latitude-E7240:~$ sudo pip3 install -U cryptography
[sudo] hasło użytkownika adasiek: 
Niestety, proszę spróbować ponownie.
[sudo] hasło użytkownika adasiek: 
Requirement already satisfied: cryptography in /usr/lib/python3/dist-packages (2.8)
Collecting cryptography
  Downloading cryptography-3.4.8-cp36-abi3-manylinux_2_24_x86_64.whl (3.0 MB)
     |████████████████████████████████| 3.0 MB 1.7 MB/s 
Collecting cffi>=1.12
  Downloading cffi-1.14.6-cp38-cp38-manylinux1_x86_64.whl (411 kB)
     |████████████████████████████████| 411 kB 2.7 MB/s 
Collecting pycparser
  Downloading pycparser-2.20-py2.py3-none-any.whl (112 kB)
     |████████████████████████████████| 112 kB 2.0 MB/s 
Installing collected packages: pycparser, cffi, cryptography
  Attempting uninstall: cryptography
    Found existing installation: cryptography 2.8
    Uninstalling cryptography-2.8:
      Successfully uninstalled cryptography-2.8
Successfully installed cffi-1.14.6 cryptography-3.4.8 pycparser-2.20
```

carlosperate, 2021-09-26:
From the log:

```
2021-09-25 10:38:40,313 - mu.virtual_environment:850(install_from_zipped_wheels) INFO: About to install from wheel: cryptography-3.4.8-cp36-abi3-manylinux_2_24_x86_64.whl
2021-09-25 10:38:40,314 - mu.virtual_environment:114(run_blocking) INFO: About to run blocking /home/adasiek/.local/share/mu/mu_venv-38-20210925-103810/bin/pip with args ['install', '--disable-pip-version-check', '--no-deps', '--no-index', '/tmp/tmphhy11gb4/cryptography-3.4.8-cp36-abi3-manylinux_2_24_x86_64.whl'] and envvars {}
2021-09-25 10:38:41,648 - mu.virtual_environment:164(wait) DEBUG: Finished: True; exitStatus 0; exitCode 1
2021-09-25 10:38:41,648 - mu.virtual_environment:185(wait) ERROR: ERROR: cryptography-3.4.8-cp36-abi3-manylinux_2_24_x86_64.whl is not a supported wheel on this platform.
2021-09-25 10:38:41,656 - mu.virtual_environment:643(ensure_and_create) ERROR: Process finished but with error code 1:
ERROR: cryptography-3.4.8-cp36-abi3-manylinux_2_24_x86_64.whl is not a supported wheel on this platform.
2021-09-25 10:38:41,656 - mu.virtual_environment:540(quarantine_venv) INFO: Quarantined /home/adasiek/.local/share/mu/mu_venv-38-20210925-103810 as /home/adasiek/.local/share/mu/mu_venv-38-20210925-103810.FAILED
2021-09-25 10:38:48,664 - root:171(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/home/adasiek/mu/mu/app.py", line 159, in run
    raise ex
  File "/home/adasiek/mu/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/adasiek/mu/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/adasiek/mu/mu/virtual_environment.py", line 768, in create
    self.install_baseline_packages()
  File "/home/adasiek/mu/mu/virtual_environment.py", line 880, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "/home/adasiek/mu/mu/virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/home/adasiek/mu/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/home/adasiek/mu/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/home/adasiek/mu/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/home/adasiek/mu/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: cryptography-3.4.8-cp36-abi3-manylinux_2_24_x86_64.whl is not a supported wheel on this platform.
```

Could you check what version of pip is installed in the virtualenv? Maybe the `manylinux_2_24` wheel format is not compatible with the version packaged with Mu, they added to cryptography in v3.4.8.

```
/home/adasiek/.local/share/mu/mu_venv-38-20210925-103810/bin/pip --version
```

abixadamj, 2021-09-26:
I tried to reproduce an error (I am on differenct machine now), remove all log and share directories, clone the actual repo.
```
Traceback (most recent call last):
  File "/home/adasiek/mu/mu/app.py", line 159, in run
    raise ex
  File "/home/adasiek/mu/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/adasiek/mu/mu/virtual_environment.py", line 627, in ensure_and_create
    self.ensure()
  File "/home/adasiek/mu/mu/virtual_environment.py", line 667, in ensure
    self.ensure_key_modules()
  File "/home/adasiek/mu/mu/virtual_environment.py", line 746, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: ipykernel
STDERR: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/adasiek/.local/share/mu/mu_venv-38-20210926-161255/lib/python3.8/site-packages/ipykernel/__init__.py", line 2, in <module>
    from .connect import *
  File "/home/adasiek/.local/share/mu/mu_venv-38-20210926-161255/lib/python3.8/site-packages/ipykernel/connect.py", line 13, in <module>
    from ipython_genutils.path import filefind
ModuleNotFoundError: No module named 'ipython_genutils'
```
[mu.log](https://github.com/mu-editor/mu/files/7231533/mu.log)

abixadamj, 2021-09-26:
But I checked:

```
adasiek@devel ~/.local/share/mu                                                                                          [16:22:34] 

./pip --version
pip 20.0.2 from /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255/lib/python3.8/site-packages/pip (python 3.8)
```

carlosperate, 2021-09-26:
Thanks @abixadamj!

The `ipython_genutils` exception is different, so I've opened a new issue to keep things organised.

Is pip at version 20.0.2 in the machine with the cryptography error?

carlosperate, 2021-09-26:
@abixadamj I think the issue might be this: https://github.com/mu-editor/mu/issues/1826

However, technically `virtualenv` should have tried to update pip on venv creation, where you offline at any point during the Mu installation and/or the first run?

I assume the Python environment where Mu was installed (either the system Python or a virtualenv) have a newer version of pip?

carlosperate, 2021-09-28:
No worries at all, we appreciate any time you can offer to test it, thanks for all the help and contributions you'd done!