mu-editor/mu #1825: Crash report - ModuleNotFoundError: No module named 'ipython_genutils'

carlosperate, 2021-09-26:
I tried to reproduce an error (I am on differenct machine now), remove all log and share directories, clone the actual repo.
```
Traceback (most recent call last):
  File "/home/adasiek/mu/mu/app.py", line 159, in run
    raise ex
  File "/home/adasiek/mu/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/adasiek/mu/mu/virtual_environment.py", line 627, in ensure_and_create
    self.ensure()
  File "/home/adasiek/mu/mu/virtual_environment.py", line 667, in ensure
    self.ensure_key_modules()
  File "/home/adasiek/mu/mu/virtual_environment.py", line 746, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: ipykernel
STDERR: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/adasiek/.local/share/mu/mu_venv-38-20210926-161255/lib/python3.8/site-packages/ipykernel/__init__.py", line 2, in <module>
    from .connect import *
  File "/home/adasiek/.local/share/mu/mu_venv-38-20210926-161255/lib/python3.8/site-packages/ipykernel/connect.py", line 13, in <module>
    from ipython_genutils.path import filefind
ModuleNotFoundError: No module named 'ipython_genutils'
```
[mu.log](https://github.com/mu-editor/mu/files/7231533/mu.log)

_Originally posted by @abixadamj in https://github.com/mu-editor/mu/issues/1823#issuecomment-927314043_

carlosperate, 2021-09-26:
Hii @abixadamj, thanks for the original report in https://github.com/mu-editor/mu/issues/1823#issuecomment-927314043!

In the machine giving you the `ModuleNotFoundError: No module named 'ipython_genutils'` could you run the following commands and share the output?

```
/home/adasiek/.local/share/mu/mu_venv-38-20210926-161255/bin/pip -c "import sys; print(sys.path)"
```
```
/home/adasiek/.local/share/mu/mu_venv-38-20210926-161255/bin/pip -m pip list --verbose
```

abixadamj, 2021-10-02:
@carlosperate - here you are ;-) I think there is only `ipython` module... 

```
adasiek@devel ~/.local/share/mu                                                                                          [16:06:17] 
> $ /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/bin/python -c "import sys; print(sys.path)"                    

['', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages']
                                                                                                                                    
adasiek@devel ~/.local/share/mu                                                                                          [16:06:33] 
> $                                                                                                                                
                                                                                                                                    
adasiek@devel ~/.local/share/mu                                                                                          [16:07:03] 
> $ /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/bin/python -m pip list --verbose                               

Package           Version Location                                                                                    Installer
----------------- ------- ------------------------------------------------------------------------------------------- ---------
backcall          0.2.0   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
bitstring         3.1.9   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
cffi              1.14.6  /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
click             8.0.1   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
cryptography      3.4.8   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
decorator         5.1.0   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
ecdsa             0.17.0  /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
entrypoints       0.3     /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
esptool           3.1     /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
Flask             1.1.2   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
ipykernel         5.5.5   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
ipython           7.28.0  /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
itsdangerous      2.0.1   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
jedi              0.18.0  /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
Jinja2            3.0.1   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
jupyter-client    7.0.3   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
jupyter-core      4.8.1   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
MarkupSafe        2.0.1   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
matplotlib-inline 0.1.3   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
nest-asyncio      1.5.1   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
numpy             1.21.2  /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
parso             0.8.2   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
pexpect           4.8.0   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
pgzero            1.2.1   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
pickleshare       0.7.5   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
pip               20.0.2  /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
pkg-resources     0.0.0   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
prompt-toolkit    3.0.20  /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
ptyprocess        0.7.0   /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
pycparser         2.20    /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip      
[... cut ...]
ipython           7.28.0  /home/adasiek/.local/share/mu/mu_venv-38-20210926-161255.FAILED/lib/python3.8/site-packages pip    
'''
On my normal system without `venv`:   

```
adasiek@devel ~/.local/share/mu                                                                                          [16:09:03] 
> $ pip3 list | grep ipyth                                                                                                         
ipython                7.13.0               
ipython-genutils       0.2.0         
```

carlosperate, 2021-10-05:
Thank you very much for the additional info @abixadamj!
I believe this turned out to be this issue, so it should be fixed in the upcoming next beta release: https://github.com/mu-editor/mu/pull/1844

closed 2021-10-05 by carlosperate

abixadamj, 2021-10-05:
OK, I will check...