mu-editor/mu #1827: Crash Report

KennQARepo, 2021-09-27:
[mu.log](https://github.com/mu-editor/mu/files/7237482/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Linux 5.4.0-42-generic #46~18.04.1-Ubuntu SMP Fri Jul 10 07:21:24 UTC 2020 x86_64

```
Traceback (most recent call last):
  File "/home/bionic/.local/lib/python3.6/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/bionic/.local/lib/python3.6/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/bionic/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/bionic/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 768, in create
    self.install_baseline_packages()
  File "/home/bionic/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 878, in install_baseline_packages
    wheels.download(zipped_wheels_filepath)
  File "/home/bionic/.local/lib/python3.6/site-packages/mu/wheels/__init__.py", line 183, in download
    pip_download(temp_dirpath, logger)
  File "/home/bionic/.local/lib/python3.6/site-packages/mu/wheels/__init__.py", line 121, in pip_download
    "Pip was unable to download %s" % pip_identifier
mu.wheels.WheelsDownloadError: Pip was unable to download esptool==3.*

```

carlosperate, 2021-09-27:
Thanks for the report @KennQARepo, what version of pip did you use to install Mu? Are you able to update pip and try again?

From the log:
```
2021-09-27 04:00:02,965 - mu.wheels:112(pip_download) DEBUG: Collecting esptool==3.*
  Downloading https://files.pythonhosted.org/packages/9c/c8/28f21b3d3b5e1f1d249be52cdd91793c8c3f7c4f4f255ece7d50984fb05d/esptool-3.1.tar.gz (175kB)
  Saved /tmp/tmp3t5irolo/esptool-3.1.tar.gz
Collecting bitstring>=3.1.6 (from esptool==3.*)
  Downloading https://files.pythonhosted.org/packages/49/fa/ac153ef3c9668a093f33386edf7a20122962e9142b1105fbe2a4a4262785/bitstring-3.1.9-py3-none-any.whl
  Saved /tmp/tmp3t5irolo/bitstring-3.1.9-py3-none-any.whl
Collecting cryptography>=2.1.4 (from esptool==3.*)
  Downloading https://files.pythonhosted.org/packages/cc/98/8a258ab4787e6f835d350639792527d2eb7946ff9fc0caca9c3f4cf5dcfe/cryptography-3.4.8.tar.gz (546kB)
  Saved /tmp/tmp3t5irolo/cryptography-3.4.8.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-qjtbtekt/cryptography/setup.py", line 14, in <module>
        from setuptools_rust import RustExtension
    ModuleNotFoundError: No module named 'setuptools_rust'
            =============================DEBUG ASSISTANCE==========================
            If you are seeing an error here please try the following to
            successfully install cryptography:
            Upgrade to the latest pip and try again. This will fix errors for most
            users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
            =============================DEBUG ASSISTANCE==========================
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-qjtbtekt/cryptography/
2021-09-27 04:00:09,977 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/home/bionic/.local/lib/python3.6/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/bionic/.local/lib/python3.6/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/bionic/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/bionic/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 768, in create
    self.install_baseline_packages()
  File "/home/bionic/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 878, in install_baseline_packages
    wheels.download(zipped_wheels_filepath)
  File "/home/bionic/.local/lib/python3.6/site-packages/mu/wheels/__init__.py", line 183, in download
    pip_download(temp_dirpath, logger)
  File "/home/bionic/.local/lib/python3.6/site-packages/mu/wheels/__init__.py", line 121, in pip_download
    "Pip was unable to download %s" % pip_identifier
mu.wheels.WheelsDownloadError: Pip was unable to download esptool==3.*
```

carlosperate, 2021-11-05:
Hi @KennQARepo, I was wondering if you had a chance to update pip and try again?

carlosperate, 2022-05-31:
Close due to lack of activity.

closed 2022-05-31 by carlosperate