mu-editor/mu #1831: Mu's packaged Python links/uses packages from other Python environments

carlosperate, 2021-09-29:


carlosperate, 2021-09-29:
@tmontes you mentioned you had some experience looking into how Python resolves paths and package locations, do you have any thoughts?

hmmkay, 2021-11-24:
Hi Carlos,

I may have a solution or workaround that involves a one-line insertion to line 783 of virtual_environment.py to add the -I switch to the python process that creates the initial virtual environment.

**My fix/workaround**
1) Install Mu as per normal and watch with sadness as it crashes when it tries to launch for the first time because it can't find the virtualenv.activation.nushell module
2) Edit C:\Users\xxxxx\AppData\Local\Programs\Mu Editor\Python\Lib\site-packages\mu\virtual_environment.py (obviously xxxxx will be your local account's name), scroll down to line 783 of the create_env method and change "-m", to "-I","-m", (ie: add the "-I", option to the python process that gets launched the first time to create the virtual environment that Mu needs. Note that -I is a capital i and not a lower-case L or the number 1). This change tells the short-lived python process that gets launched to do some first-time setups of Mu's virtual python environment should run in isolated mode and not use system packages
3) Save the file and try start Mu again.

For me this resolved the issue and Mu finally launched

Whether this is the correct solution and free of future consequences down the road I don't know. But it helped me get Mu actually running and might be helpful to some fellow sufferer out there!

**Background details**

I was experiencing the same crashes (No module named 'virtualenv.activation.nushell') that completely prevented me launching Mu 1.1.0.beta.6 at all on Windows, and despite several clean uninstall/re-install attempts.

I followed the breadcrumb trail you left of looking at the sys.path, and saw the same issue you described above that my system-wide Python's site-packages directory 'C:\\Users\\xxxxx\\AppData\\Roaming\\Python\\Python38\\site-packages' was included in the sys.path, and that also it appeared before the site-packages of the python bundled by Mu-Editor.

Checking my filesystem, I saw that my system-wide, C:\Users\xxxxx\AppData\Roaming\Python\Python38\site-packages includes the virtual virtualenv.activation package, but not virtualenv.activation.nushell.

To confirm that Mu's bundled python was actually accessing my system-wide python packages I launched Mu'd bundled python, imported virtualenv.activation and saw that the __file__ property of the package referenced the wrong version
```
C:\Users\xxxxx\AppData\Local\Programs\Mu Editor\Python>.\python
Python 3.8.11 (default, Jul 24 2021, 23:33:17) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import virtualenv.activation
>>> virtualenv.activation.__file__
'C:\\Users\\xxxxx\\AppData\\Roaming\\Python\\Python38\\site-packages\\virtualenv\\activation\\__init__.py'
```

... but you can actually make the module import work by manually inserting the correct site-packages path to Mu's bundled python into sys.path:
```
C:\Users\xxxxx\AppData\Local\Programs\Mu Editor\Python>.\python
Python 3.8.11 (default, Jul 24 2021, 23:33:17) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.path.insert(1, 'c:\\Users\\xxxxx\\AppData\\Local\\Programs\\Mu Editor\\Python\\Lib\\site-packages')
>>> import virtualenv.activation.nushell
>>> virtualenv.activation.nushell.__file__
'c:\\Users\\xxxxx\\AppData\\Local\\Programs\\Mu Editor\\Python\\Lib\\site-packages\\virtualenv\\activation\\nushell\\__init__.py'
```

... insert French accented several-hours-later meme ...

Much reading about packagers and installer later, I was reading through Mu's own packaging docs at https://mu.readthedocs.io/en/latest/packaging.html and noticed that the bundled Python version runs in isolated mode.

The  link to Python's doc on isolated mode indicates that this mode isolates the Python process from the user's site-packages directory and other system settings. So why then do we see our system-wide site-packages in sys.path?

My reading of Python's docs leads me to believe that isolated mode is only activated with the -I command line option, and I don't see that option included in the create_venv() method of virtual_environment.py.

So trying out the -I option from the command line shows that we can now import the modules without error using Mu's bundled Python!
```
C:\Users\xxxxx\AppData\Local\Programs\Mu Editor\Python>.\python -I
Python 3.8.11 (default, Jul 24 2021, 23:33:17) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import virtualenv.activation.nushell
>>> virtualenv.activation.__file__
'C:\\Users\\xxxxx\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\virtualenv\\activation\\__init__.py'
```

hmmkay, 2021-11-25:
@2low2free: Thanks for taking the time to let me know. I'm very pleased that my little contribution has helped somebody else!

carlosperate, 2021-12-06:
Wow, that's a great find @hmmkay, thanks!
Would you be able to send us a PR with the change, that would automatically generate test installers that people could use for testing.

tjguk, 2021-12-13:
Thanks from me, too, @hmmkay (as the maintainer of the venv management within Mu).

I can't see that you've submitted a PR so I'll create one myself and credit you for the code change. Hopefully this will help others fix the same problem

carlosperate, 2021-12-15:
Thanks again @hmmkay! With @tjguk PR we can start testing this installer 🎉 

@2low2free would you also be able to try the installer at the bottom of this page, in the "Artifacts" section to see if it resolves the problem?
https://github.com/mu-editor/mu/actions/runs/1574840282

To ensure things go smoothly it'd be highly recommended to first uninstall the version of Mu in your PC before installing this version.

hmmkay, 2021-12-16:
Sorry to have gone quiet on you and not done a PR as requested - but looks like you guys have everything in hand and are forging ahead! Noice!

carlosperate, 2021-12-16:
No worries at all, we really appreciate you spent the time to debug and fix the issue, thanks again!

carlosperate, 2021-12-27:
Okay so we've got a couple of findings

1. Some users still experience this issue
     - https://github.com/mu-editor/mu/issues/1602#issuecomment-1000907359
     - This is because the fix was applied to Mu launching the Python interpreter to create a venv, but some people had issues to launch Mu at all because some of the "core packages" were being picked up by the user environment instead of the Mu bundled site-packages
     - Launching Mu with the `-I` flag as well resolves this
     - @tmontes This would need to be fixed in pup: https://github.com/mu-editor/mu/issues/1954  
2. Debian/Ubuntu issues
    - https://github.com/mu-editor/mu/issues/1954
    - Debian pip installs packages in the "user environment" and the fix to this issue isolates the interpreter from the "user environment", so for people that installed Mu not in a venv it mens it cannot find the virtualenv module

tmontes, 2022-03-21:
> Okay so we've got a couple of findings
> 
> 1. Some users still experience this issue
>    
>    * [Mu 1.1.0-beta-4 wont run at all #1602 (comment)](https://github.com/mu-editor/mu/issues/1602#issuecomment-1000907359)
>    * [Crash Report #1955](https://github.com/mu-editor/mu/issues/1955)
>    * This is because the fix was applied to Mu launching the Python interpreter to create a venv, but some people had issues to launch Mu at all because some of the "core packages" were being picked up by the user environment instead of the Mu bundled site-packages
>    * Launching Mu with the `-I` flag as well resolves this
>    * @tmontes This would need to be fixed in pup: [Issue launching virtualenv module isolated from user's environment #1954](https://github.com/mu-editor/mu/issues/1954)

...just released `pup` 1.0.0a15 that defaults to launching with the `-I` flag.

carlosperate, 2022-06-03:
The same kind of problem was also ocurrying when installing jupyter inside the venv, so PR https://github.com/mu-editor/mu/pull/2237 should have fixed that as well.

closed 2022-06-09 by carlosperate