mu-editor/mu #1837: Crash Report

Rodrigo5373, 2021-10-01:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: es_SV
Platform: Windows 10 10.0.19041 AMD64

```
all last):
  File "C:\Users\alebo\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\alebo\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\alebo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\alebo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\alebo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 365, in run
    editor.restore_session(sys.argv[1:])
  File "C:\Users\alebo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 888, in restore_session
    self.direct_load(old_path)
  File "C:\Users\alebo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 1145, in direct_load
    self._load(path)
  File "C:\Users\alebo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 1017, in _load
    text, newline = read_and_decode(path)
  File "C:\Users\alebo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 318, in read_and_decode
    text = btext.decode(encoding)
LookupError: unknown encoding: uft-8

```

carlosperate, 2021-10-01:
Thanks for the report !
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\alebo\AppData\Local\python\mu\Logs\mu.log`

And after attaching the logs, could you download, install and try the installer from this link? https://github.com/mu-editor/mu/actions/runs/1266857664
It's named `mu-editor-Windows-0a954b6da7f4ccc4c8f3519bdfc3f2e49c0589d8` at the bottom of the page.

I think that version might resolve this issue.

carlosperate, 2021-10-04:
Thank you for the report @Rodrigo5373, this should be fixed in the next beta release of Mu. 👍

closed 2021-10-04 by carlosperate