mu-editor/mu #1840: Crash Report

ntoll, 2021-10-03:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: en_GB
Platform: Darwin 19.5.0 Darwin Kernel Version 19.5.0: Tue May 26 20:41:44 PDT 2020; root:xnu-6153.121.2~2/RELEASE_X86_64 x86_64

```
on3.8/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 627, in ensure_and_create
    self.ensure()
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 667, in ensure
    self.ensure_key_modules()
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 746, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: ipykernel
STDERR: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/ntoll/Library/Application Support/mu/mu_venv-38-20211003-151335/lib/python3.8/site-packages/ipykernel/__init__.py", line 2, in <module>
    from .connect import *
  File "/Users/ntoll/Library/Application Support/mu/mu_venv-38-20211003-151335/lib/python3.8/site-packages/ipykernel/connect.py", line 13, in <module>
    from ipython_genutils.path import filefind
ModuleNotFoundError: No module named 'ipython_genutils'

``
[mu.log](https://github.com/mu-editor/mu/files/7273479/mu.log)
`

ntoll, 2021-10-03:
I've also deleted the mu related directories and tried this from "clean". Same problem.

@carlosperate I seem to remember you were fixing some things related to iPython..? I suspect a package version needs pinning to something else, or perhaps pup has gone wrong and missed a package..? @tmontes any thoughts?

ntoll, 2021-10-03:
I see this in the build output from pup:

```
pip out: Successfully installed PyQt5-5.13.2 PyQt5-sip-12.9.0 PyQtChart-5.13.1 QScintilla-2.11.3 adafruit-board-toolkit-1.1.0 appdirs-1.4.4 appnope-0.1.2 backcall-0.2.0 backports.entry-points-selectable-1.1.0 black-21.9b0 click-8.0.1 decorator-5.1.0 distlib-0.3.3 filelock-3.3.0 flake8-3.9.2 ipykernel-5.5.5 ipython-7.28.0 ipython-genutils-0.2.0 jedi-0.18.0 jupyter-client-6.1.12 jupyter-core-4.8.1 matplotlib-inline-0.1.3 mccabe-0.6.1 mu-editor-1.1.0b6 mypy-extensions-0.4.3 nudatus-0.0.5 parso-0.8.2 pathspec-0.9.0 pexpect-4.8.0 pickleshare-0.7.5 platformdirs-2.4.0 prompt-toolkit-3.0.20 ptyprocess-0.7.0 pycodestyle-2.7.0 pyflakes-2.3.1 pygments-2.10.0 pyserial-3.5 python-dateutil-2.8.2 pyzmq-22.3.0 qtconsole-4.7.7 qtpy-1.11.2 regex-2021.9.30 semver-2.13.0 six-1.16.0 tomli-1.2.1 tornado-6.1 traitlets-5.1.0 typing-extensions-3.10.0.2 virtualenv-20.8.1 wcwidth-0.2.5 wheel-0.37.0
```

Note the presence of `ipython-genutils-0.2.0` (the package that's allegedly missing).

ntoll, 2021-10-03:
@tmontes is pup missing something here..?

tmontes, 2021-10-03:
My experience:

* Updated local mu repo to latest master.
* Recretated 3.8 venv and `pip install -e .` into it.
* Removed any previously existing mu venv and settings.
* Launching mu fails with the same exact error -- does not seem packaging related.

What I tried:

* Added `ipython_genutils` to the required packages in Mu's execution venv:
```
(venv38) tiago.montes (~/Work/Python/github.com/mu): git diff
diff --git a/mu/wheels/__init__.py b/mu/wheels/__init__.py
index 04ade70..0a06071 100644
--- a/mu/wheels/__init__.py
+++ b/mu/wheels/__init__.py
@@ -45,6 +45,7 @@ mode_packages = [
     # FIXME: ipykernel max ver added for macOS 10.13 compatibility, min taken
     # from qtconsole 4.7.7. This is mirrored in setup.py
     ("ipykernel", "ipykernel>=4.1,<6"),
+    ("ipython_genutils", "ipython_genutils==0.2.0"),^M
     ("esptool", "esptool==3.*"),
 ]
```

* Mu now launches.

tmontes, 2021-10-03:
More:

* The `pup`-packaged version seems to run fine, too, with the above listed fix.

tmontes, 2021-10-03:
LINKING to Windows failure in #1841:

* Behaves the same way and fixed it the same way. :)

carlosperate, 2021-10-04:
This looks related: https://github.com/ipython/ipykernel/issues/759

carlosperate, 2021-10-04:
And we are using `ipykernel>=4.1,<6` to be compatible with old macos and to be in sync with the `qtconsole==4.7.7` requirement:
https://github.com/mu-editor/mu/blob/3f072a968bb59414e1a1e8de2915044f6016ea34/setup.py#L37-L41

So it looks like this is indeed the problem.

closed 2021-10-05 by carlosperate