mu-editor/mu #1841: Crash Report

ntoll, 2021-10-03:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: en_GB
Platform: Windows 10 10.0.19041 AMD64

```
st):
  File "C:\Users\ntoll\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\ntoll\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\ntoll\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 627, in ensure_and_create
    self.ensure()
  File "C:\Users\ntoll\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 667, in ensure
    self.ensure_key_modules()
  File "C:\Users\ntoll\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 746, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: ipykernel
STDERR: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\ntoll\AppData\Local\python\mu\mu_venv-38-20211003-163325\lib\site-packages\ipykernel\__init__.py", line 2, in <module>
    from .connect import *
  File "C:\Users\ntoll\AppData\Local\python\mu\mu_venv-38-20211003-163325\lib\site-packages\ipykernel\connect.py", line 13, in <module>
    from ipython_genutils.path import filefind
ModuleNotFoundError: No module named 'ipython_genutils'

```

Related to #1840 

[mu.log](https://github.com/mu-editor/mu/files/7273849/mu.log)

tmontes, 2021-10-03:
Chiming in to confirm that my experience in macOS, documented in #1840, was bit-by-bit replicated on Windows:

* Cannot run from source: fails in the same way.
* Hacking `mu/wheels/__init__.py` fixes it and:
  * Runs from source.
  * Runs after `pup`-packaging.

carlosperate, 2021-10-05:
As this is a platform independent issue I'll close this one to continue the conversation over https://github.com/mu-editor/mu/issues/1840.

closed 2021-10-05 by carlosperate