mu-editor/mu #1843: Document Windows installer advanced options

carlosperate, 2021-10-04:


tmontes, 2021-11-04:
Docs are here:

* https://pup.readthedocs.io/en/latest/platform_notes/windows_msi.html#per-user-vs-per-machine-installs

I say we should "copy+paste" the linked section into Mu's own docs: readers will probably be confused about `pup` and the remaining information on that page! ;)

carlosperate, 2022-10-07:
@tmontes is there an extra quote mark in the command?

```
$ msiexec /i <msi-file>" MSIINSTALLPERUSER=""
                       △
                       |
```