mu-editor/mu #1853: Crash Report

longyuan94, 2021-10-14:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: zh_CN
Platform: Windows 10 10.0.19042 AMD64

```
Traceback (most recent call last):
  File "C:\Users\leolong\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\leolong\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\leolong\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 627, in ensure_and_create
    self.ensure()
  File "C:\Users\leolong\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 665, in ensure
    self.ensure_interpreter_version()
  File "C:\Users\leolong\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 731, in ensure_interpreter_version
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Mu interpreter at version 38; venv interpreter at version 38
[0x7FFB3D10E0A4] ANOMALY: use of REX.w is meaningless (default operand size is 64)
[0x7FFB3D10E0A4] ANOMALY: use of REX.w is meaningless (default operand size is 64).

```

carlosperate, 2021-10-14:
Thanks for the report @longyuan94!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\leolong\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-12-17:
Hi @longyuan94, are you still experiencing this problem?

Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: C:\Users\leolong\AppData\Local\python\mu\Logs\mu.log

carlosperate, 2022-03-07:
The link here might be useful: https://github.com/microsoft/terminal/issues/6391#issuecomment-695985986

Closing due to lack of activity.

closed 2022-03-07 by carlosperate