mu-editor/mu #1862: Crash Report

kylongmu, 2021-10-21:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: zh_CN
Platform: Windows 10 10.0.19043 AMD64

```
Traceback (most recent call last):
  File "C:\Users\THINKPAD\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\THINKPAD\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\THINKPAD\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Users\THINKPAD\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 767, in create
    self.create_venv()
  File "C:\Users\THINKPAD\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 798, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\THINKPAD\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\THINKPAD\AppData\Local\python\mu\mu_venv-38-20211022-000243
ModuleNotFoundError: No module named 'virtualenv.seed.embed.via_app_data'

```

kylongmu, 2021-10-21:
> Thanks for the report @kylongmu ! Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\THINKPAD\AppData\Local\python\mu\Logs\mu.log`
[mu.log](https://github.com/mu-editor/mu/files/7391613/mu.log)

carlosperate, 2021-10-21:
Thanks @kylongmu !

I think you might be encountering this issue: https://github.com/mu-editor/mu/issues/1831

Could you run this command in a Command Prompt and paste the result here?

```
"C:\Users\THINKPAD\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
```

carlosperate, 2021-12-15:
@kylongmu We think we might have found a solution. Would you be able to try the installer at the bottom of this page, in the "Artifacts" section to see if it resolves the problem you are encountering?
https://github.com/mu-editor/mu/actions/runs/1574840282

To ensure things go smoothly it'd be highly recommended to first uninstall the version of Mu in your PC before installing this version.

Getting your confirmation would be very useful to us to confirm the issue has been fixed.