mu-editor/mu #1871: Crash Report

hazelbooth, 2021-10-31:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: en_US
Platform: Windows 10 10.0.18362 AMD64

```
   raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\24819\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\24819\AppData\Local\python\mu\mu_venv-38-20211031-103249
RuntimeError: failed to build image pip because:
Traceback (most recent call last):
  File "C:\Users\24819\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\via_app_data.py", line 57, in _install
    installer.install(creator.interpreter.version_info)
  File "C:\Users\24819\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 35, in install
    self._uninstall_previous_version()
  File "C:\Users\24819\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 151, in _uninstall_previous_version
    dist_name = self._dist_info.stem.split("-")[0]
  File "C:\Users\24819\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 115, in _dist_info
    raise RuntimeError(msg)  # pragma: no cover
RuntimeError: no .dist-info at D:\Users\24819\AppData\Local\pypa\virtualenv\wheel\3.8\image\1\CopyPipInstall\pip-21.2.4-py3-none-any, has pip

```

hazelbooth, 2021-10-31:
Starting Mu 1.1.0.beta.6
2021-10-31 10:32:41,589 - root:270(run) INFO: uname_result(system='Windows', node='LS00966', release='10', version='10.0.18362', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 12, GenuineIntel')
2021-10-31 10:32:41,652 - root:271(run) INFO: Platform: Windows-10-10.0.18362-SP0
2021-10-31 10:32:41,652 - root:272(run) INFO: Python path: ['C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-57.4.0-py3.8.egg', 'C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-21.1.3-py3.8.egg', 'C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-10-31 10:32:41,653 - root:273(run) INFO: Language code: en_US
2021-10-31 10:32:41,653 - mu.settings:220(load) WARNING: No settings file found at C:\Users\24819\AppData\Local\python\mu\settings.json; skipping
2021-10-31 10:32:42,294 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2021-10-31 10:32:42,295 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-10-31 10:32:42,296 - mu.virtual_environment:777(create_venv) INFO: Creating virtualenv: C:\Users\24819\AppData\Local\python\mu\mu_venv-38-20211029-193402
2021-10-31 10:32:42,296 - mu.virtual_environment:778(create_venv) INFO: Virtualenv name: mu_venv-38-20211029-193402
2021-10-31 10:32:42,296 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-m', 'virtualenv', '-p', 'C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-q', 'C:\\Users\\24819\\AppData\\Local\\python\\mu\\mu_venv-38-20211029-193402') with kwargs {'env': {'ALLUSERSPROFILE': 'C:\\ProgramData', 'APPDATA': 'C:\\Users\\24819\\AppData\\Roaming', 'COMMONPROGRAMFILES': 'C:\\Program Files\\Common Files', 'COMMONPROGRAMFILES(X86)': 'C:\\Program Files (x86)\\Common Files', 'COMMONPROGRAMW6432': 'C:\\Program Files\\Common Files', 'COMPUTERNAME': 'LS00966', 'COMSPEC': 'C:\\WINDOWS\\system32\\cmd.exe', 'DRIVERDATA': 'C:\\Windows\\System32\\Drivers\\DriverData', 'FPS_BROWSER_APP_PROFILE_STRING': 'Internet Explorer', 'FPS_BROWSER_USER_PROFILE_STRING': 'Default', 'HOMEDRIVE': 'C:', 'HOMEPATH': '\\Users\\24819', 'LOCALAPPDATA': 'C:\\Users\\24819\\AppData\\Local', 'LOGONSERVER': '\\\\SHAD04', 'NUMBER_OF_PROCESSORS': '8', 'ONEDRIVE': 'C:\\Users\\24819\\OneDrive - somerville.qld.edu.au', 'ONEDRIVECOMMERCIAL': 'C:\\Users\\24819\\OneDrive - somerville.qld.edu.au', 'OS': 'Windows_NT', 'PATH': 'C:\\Users\\24819\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\PyQt5\\Qt\\bin;C:\\Program Files (x86)\\Common Files\\Oracle\\Java\\javapath;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Program Files (x86)\\Windows Live\\Shared;C:\\WINDOWS\\CCM;C:\\WINDOWS\\CCM;C:\\WINDOWS\\CCM;C:\\WINDOWS\\CCM;C:\\WINDOWS\\CCM;C:\\WINDOWS\\CCM;C:\\WINDOWS\\CCM;C:\\WINDOWS\\CCM;C:\\WINDOWS\\CCM;C:\\WINDOWS\\CCM;C:\\WINDOWS\\CCM;C:\\WINDOWS\\CCM;C:\\WINDOWS\\CCM;C:\\Users\\24819\\AppData\\Local\\Microsoft\\WindowsApps;', 'PATHEXT': '.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC', 'PROCESSOR_ARCHITECTURE': 'AMD64', 'PROCESSOR_IDENTIFIER': 'Intel64 Family 6 Model 142 Stepping 12, GenuineIntel', 'PROCESSOR_LEVEL': '6', 'PROCESSOR_REVISION': '8e0c', 'PROGRAMDATA': 'C:\\ProgramData', 'PROGRAMFILES': 'C:\\Program Files', 'PROGRAMFILES(X86)': 'C:\\Program Files (x86)', 'PROGRAMW6432': 'C:\\Program Files', 'PSMODULEPATH': 'C:\\Program Files\\WindowsPowerShell\\Modules;C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules', 'PUBLIC': 'C:\\Users\\Public', 'SESSIONNAME': 'Console', 'SYSTEMDRIVE': 'C:', 'SYSTEMROOT': 'C:\\WINDOWS', 'TEMP': 'C:\\Users\\24819\\AppData\\Local\\Temp', 'TMP': 'C:\\Users\\24819\\AppData\\Local\\Temp', 'UATDATA': 'C:\\WINDOWS\\CCM\\UATData\\D9F8C395-CAB8-491d-B8AC-179A1FE1BE77', 'USERDNSDOMAIN': 'SOMERVILLE.QLD.EDU.AU', 'USERDOMAIN': 'SH', 'USERDOMAIN_ROAMINGPROFILE': 'SH', 'USERNAME': '24819', 'USERPROFILE': 'C:\\Users\\24819', 'WINDIR': 'C:\\WINDOWS', 'QT_API': 'pyqt5', 'QT_AUTO_SCREEN_SCALE_FACTOR': '1', 'QT_MAC_WANTS_LAYER': '1'}}
2021-10-31 10:32:49,426 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 1; output: RuntimeError: failed to build image pip because:
Traceback (most recent call last):
  File "C:\Users\24819\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\via_app_data.py", line 57, in _install
    installer.install(creator.interpreter.version_info)
[... cut ...]
  File "C:\Users\24819\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 115, in _dist_info
    raise RuntimeError(msg)  # pragma: no cover
RuntimeError: no .dist-info at D:\Users\24819\AppData\Local\pypa\virtualenv\wheel\3.8\image\1\CopyPipInstall\pip-21.2.4-py3-none-any, has pip
2021-10-31 10:32:58,887 - mu.settings:169(save) DEBUG: Saving to C:\Users\24819\AppData\Local\python\mu\session.json
2021-10-31 10:32:58,888 - mu.settings:169(save) DEBUG: Saving to C:\Users\24819\AppData\Local\python\mu\venv.json

hazelbooth, 2021-10-31:
that was my log file please help me i need mu before monday

carlosperate, 2021-11-02:
This looks the same issue as: https://github.com/mu-editor/mu/issues/1820
A bit of info in: https://github.com/mu-editor/mu/issues/1820#issuecomment-929030978

@tjguk should we ask the user to try to reset the virtualenv app data as shown in https://github.com/pypa/virtualenv/issues/1971, or is there anything else you'd like to check first?