mu-editor/mu #1875: Crash Report

ivan-yosifov, 2021-11-02:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: en_US
Platform: Windows 10 10.0.19043 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Ivan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\Ivan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Ivan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Users\Ivan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 767, in create
    self.create_venv()
  File "C:\Users\Ivan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 798, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\Ivan\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\Ivan\AppData\Local\python\mu\mu_venv-38-20211102-115214
ModuleNotFoundError: No module named 'virtualenv.activation.nushell'

```

carlosperate, 2021-11-02:
Thanks for the report @ivan-yosifov!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\Ivan\AppData\Local\python\mu\Logs\mu.log`

Could you also run the two commands in a command prompt and copy the result here? It would be really helpful:

```
"C:\Users\Ivan\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
```
```
"C:\Users\Ivan\AppData\Local\python\mu\mu_venv-38-20211102-115214\scripts\python.exe" -m pip list --verbose
```

ivan-yosifov, 2021-11-02:
This is the log
https://drive.google.com/file/d/1wRX3k7-OjqH2iiy9WNBJrwE0xuTcjuTH/view?usp=sharing

ivan-yosifov, 2021-11-02:
The first command returned the following:
[first.txt](https://github.com/mu-editor/mu/files/7459754/first.txt)

ivan-yosifov, 2021-11-02:
The second command returned
The system cannot find the path specified.

carlosperate, 2021-11-02:
Thanks @ivan-yosifov!

We can see from the command that worked that these packages are being picked up from the global Python environment and not the packaged one:

```
Package                           Version   Location                                                                                            Installer
--------------------------------- --------- --------------------------------------------------------------------------------------------------- ---------
appdirs                           1.4.4     c:\users\ivan\appdata\roaming\python\python38\site-packages                                         pip
certifi                           2020.6.20 c:\users\ivan\appdata\roaming\python\python38\site-packages                                         pip
distlib                           0.3.1     c:\users\ivan\appdata\roaming\python\python38\site-packages                                         pip
filelock                          3.0.12    c:\users\ivan\appdata\roaming\python\python38\site-packages                                         pip
pipenv                            2020.6.2  c:\users\ivan\appdata\roaming\python\python38\site-packages                                         pip
six                               1.15.0    c:\users\ivan\appdata\roaming\python\python38\site-packages                                         pip
virtualenv                        20.0.26   c:\users\ivan\appdata\roaming\python\python38\site-packages                                         pip
virtualenv-clone                  0.5.4     c:\users\ivan\appdata\roaming\python\python38\site-packages                                         pip
```

That includes `virtualenv`, which is likely at a different version than the Mu package and would explain the error.
So it looks like the problem is caused by this: https://github.com/mu-editor/mu/issues/1831

We are currently investigating this, thanks for the report!

ivan-yosifov, 2021-11-02:
@carlosperate I reinstalled the Mu editor and still get the same error. 
Is there a way to start clean without any packages being loaded?

carlosperate, 2021-12-15:
@ivan-yosifov We think we might have found a solution. Would you be able to try the installer at the bottom of this page, in the "Artifacts" section to see if it resolves the problem you are encountering?
https://github.com/mu-editor/mu/actions/runs/1574840282

To ensure things go smoothly it'd be highly recommended to first uninstall the version of Mu in your PC before installing this version.

If you could try this installer it would be very very useful to us to confirm the issue has been fixed.

ivan-yosifov, 2021-12-18:
Hello Carlos,Unfortunately I can not test it out because I finished work for this year. And this problem was on my work computer. 
I am going back to work on Jan 10, 2022 so I'll test it out immediately and let you know.
Thank you for your hard workIvan

 
 

https://github.com/mu-editor/mu/actions/runs/1574840282

carlosperate, 2021-12-18:
Hi Ivan,

Thanks for getting back to us! If you could test it when you are back that'd be great, have a great holiday!

carlosperate, 2022-03-07:
Hi @ivan-yosifov, I was wondering if you had a chance to test this with the latest release? https://github.com/mu-editor/mu/releases/tag/v1.1.1

ivan-yosifov, 2022-03-08:
Hi Carlos,
we completely moved to home office at the beginning of this year. Later this month I'll be going to the office and test it out. 
I'll let you know how it went.
Thank you,Ivan

carlosperate, 2022-03-08:
Sounds good, hank you @ivan-yosifov!

ivan-yosifov, 2022-04-15:
Hi Carlos,good news. Just tested the latest release 1.1.1. Works great. No crashes. 

Thank you for your patienceIvan

carlosperate, 2022-04-19:
That's great, thank you for testing again and the update @ivan-yosifov!

closed 2022-04-19 by carlosperate