mu-editor/mu #1876: Crash Report

tasnimmm, 2021-11-03:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: fr_FR
Platform: Windows 10 10.0.19043 AMD64

```
31, in cleanup
    self._rmtree(self.name)
  File "C:\Users\tasso\AppData\Local\Programs\Mu Editor\Python\lib\tempfile.py", line 813, in _rmtree
    _shutil.rmtree(name, onerror=onerror)
  File "C:\Users\tasso\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 740, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\tasso\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 618, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\tasso\AppData\Local\Programs\Mu Editor\Python\lib\tempfile.py", line 805, in onerror
    cls._rmtree(path)
  File "C:\Users\tasso\AppData\Local\Programs\Mu Editor\Python\lib\tempfile.py", line 813, in _rmtree
    _shutil.rmtree(name, onerror=onerror)
  File "C:\Users\tasso\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 740, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\tasso\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 599, in _rmtree_unsafe
    onerror(os.scandir, path, sys.exc_info())
  File "C:\Users\tasso\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 596, in _rmtree_unsafe
    with os.scandir(path) as scandir_it:
NotADirectoryError: [WinError 267] Nom de rÃ©pertoire non valide: 'C:\\Users\\tasso\\AppData\\Local\\Temp\\tmpfwn9181v\\pygame-2.0.1-cp38-cp38-win_amd64.whl'

```

carlosperate, 2021-11-03:
Thanks for the report @tasnimmm!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\tasso\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-12-17:
Hi @tasnimmm, are you still experiencing this problem?

Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: C:\Users\tasso\AppData\Local\python\mu\Logs\mu.log

carlosperate, 2022-05-31:
Close due to lack of activity.

closed 2022-05-31 by carlosperate