mu-editor/mu #1881: Crash Report

AnnaBp908, 2021-11-05:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: ro_RO
Platform: Windows 10 10.0.18363 AMD64

```
Traceback (most recent call last):
  File "C:\Users\matei\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\matei\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\matei\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\matei\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\matei\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 349, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\matei\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 809, in setup
    os.makedirs(wd)
  File "C:\Users\matei\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\matei\\mu_code'

```

carlosperate, 2021-11-09:
Thanks for the report @AnnaBp908!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\matei\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-12-17:
Hi @AnnaBp908, I was wondering if you are still experiencing this issue?
If so, could you attach the logs to this GitHub issue? File should be located in: `C:\Users\matei\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2022-03-07:
Hi @AnnaBp908o, I suspect this problem might have been caused by https://github.com/mu-editor/mu/issues/1755.

Are you still experiencing this error? Would you be able to find the logs and send them to us? File should be located in: C:\Users\matei\AppData\Local\python\mu\Logs\mu.log

carlosperate, 2022-05-31:
Close due to lack of activity.

closed 2022-05-31 by carlosperate