mu-editor/mu #1884: Crash Report

MOARAGON, 2021-11-09:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: en_US
Platform: Linux 5.4.139-16309-g748e5311576e #1 SMP PREEMPT Sun Oct 17 21:42:00 PDT 2021 aarch64

```
Traceback (most recent call last):
  File "/home/moisesaragon96/.local/lib/python3.7/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/moisesaragon96/.local/lib/python3.7/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/moisesaragon96/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/moisesaragon96/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 768, in create
    self.install_baseline_packages()
  File "/home/moisesaragon96/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 878, in install_baseline_packages
    wheels.download(zipped_wheels_filepath, logger)
  File "/home/moisesaragon96/.local/lib/python3.7/site-packages/mu/wheels/__init__.py", line 192, in download
    pip_download(temp_dirpath, logger)
  File "/home/moisesaragon96/.local/lib/python3.7/site-packages/mu/wheels/__init__.py", line 130, in pip_download
    "Pip was unable to download %s" % pip_identifier
mu.wheels.WheelsDownloadError: Pip was unable to download pgzero>=1.2.1

```

carlosperate, 2021-11-09:
Thanks for the report @MOARAGON!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

MOARAGON, 2021-11-10:
Hi @carlosperate, I'm unfamiliar with these processes. I've tried searching for the log file but I keep hitting walls such as: "You're file could not be acessed". Attached below is what I received from the crash report. 
/home/moisesaragon96/.cache/mu/log/mu.log
Is there anyway I can help to resolve this issue?

carlosperate, 2021-11-10:
Are you able to open the log file and copy the contents here?
If it gives you an access error it might be that you've installed and/or run Mu with `sudo`?

Do you get the same error if you open Mu with `sudo`?

carlosperate, 2022-03-07:
Hi @MOARAGON, I was wondering if you were still encountering this issue?

carlosperate, 2022-05-31:
Close due to lack of activity.
@MOARAGON if you are still experiencing this issue is possible you might need to use `sudo` to open Mu. If that still doesn't work please do reopen this issue so that we can look for a solution.

closed 2022-05-31 by carlosperate