mu-editor/mu #1887: Crash Report

7Wood, 2021-11-13:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: en_US
Platform: Linux 5.4.139-16311-g330b3df0010b #1 SMP PREEMPT Mon Nov 8 12:23:05 PST 2021 x86_64

```
Traceback (most recent call last):
  File "/home/lynnwoods/.local/lib/python3.7/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/lynnwoods/.local/lib/python3.7/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/lynnwoods/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/lynnwoods/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 768, in create
    self.install_baseline_packages()
  File "/home/lynnwoods/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 878, in install_baseline_packages
    wheels.download(zipped_wheels_filepath, logger)
  File "/home/lynnwoods/.local/lib/python3.7/site-packages/mu/wheels/__init__.py", line 192, in download
    pip_download(temp_dirpath, logger)
  File "/home/lynnwoods/.local/lib/python3.7/site-packages/mu/wheels/__init__.py", line 130, in pip_download
    "Pip was unable to download %s" % pip_identifier
mu.wheels.WheelsDownloadError: Pip was unable to download esptool==3.*

```

carlosperate, 2021-11-22:
Thanks for the report @7Wood!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

carlosperate, 2021-12-17:
Hi @7Wood, are you still experiencing this problem?
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

carlosperate, 2022-05-31:
Close due to lack of activity.

7Wood, 2022-05-31:
I have resolved it.  

Thank you.

Lynnwood

carlosperate, 2022-05-31:
Oh that's great! Thanks for the update 👍

7Wood, 2022-05-31:
Thank YOU for following up.

Lynnwood