mu-editor/mu #1892: Crash Report

MINIbra1n, 2021-11-20:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: ru_RU
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\hedge\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\hedge\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\hedge\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\hedge\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\hedge\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 349, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\hedge\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 809, in setup
    os.makedirs(wd)
  File "C:\Users\hedge\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] ÐÐµÐ²Ð¾Ð·Ð¼Ð¾Ð¶Ð½Ð¾ ÑÐ¾Ð·Ð´Ð°ÑÑ ÑÐ°Ð¹Ð», ÑÐ°Ðº ÐºÐ°Ðº Ð¾Ð½ ÑÐ¶Ðµ ÑÑÑÐµÑÑÐ²ÑÐµÑ: 'C:\\Users\\hedge\\mu_code'

```

carlosperate, 2021-12-17:
Hi @MINIbra1n, I suspect this problem might have been caused by https://github.com/mu-editor/mu/issues/1755.

If you are still experiencing this error, are you able to find the logs and send them to us? File should be located in: `C:\Users\hedge\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2022-03-07:
Hi @MINIbra1n, are you still encountering this issue?

carlosperate, 2022-05-31:
Close due to lack of activity.