mu-editor/mu #1893: Crash Report

mouhib53, 2021-11-20:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: fr_FR
Platform: Windows 10 10.0.19044 AMD64

```
, line 831, in cleanup
    self._rmtree(self.name)
  File "C:\Users\asus\AppData\Local\Programs\Mu Editor\Python\lib\tempfile.py", line 813, in _rmtree
    _shutil.rmtree(name, onerror=onerror)
  File "C:\Users\asus\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 740, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\asus\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 618, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\asus\AppData\Local\Programs\Mu Editor\Python\lib\tempfile.py", line 805, in onerror
    cls._rmtree(path)
  File "C:\Users\asus\AppData\Local\Programs\Mu Editor\Python\lib\tempfile.py", line 813, in _rmtree
    _shutil.rmtree(name, onerror=onerror)
  File "C:\Users\asus\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 740, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\asus\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 599, in _rmtree_unsafe
    onerror(os.scandir, path, sys.exc_info())
  File "C:\Users\asus\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 596, in _rmtree_unsafe
    with os.scandir(path) as scandir_it:
NotADirectoryError: [WinError 267] Nom de rÃ©pertoire non valide: 'C:\\Users\\asus\\AppData\\Local\\Temp\\tmp8ux55oba\\numpy-1.21.2-cp38-cp38-win_amd64.whl'

```

carlosperate, 2021-11-22:
Thanks for the report @mouhib53!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\asus\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-12-17:
Hi @mouhib53, are you still experiencing this problem?
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\asus\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2022-05-31:
Close due to lack of activity.

closed 2022-05-31 by carlosperate