mu-editor/mu #1894: Can't open esp32s file system

SpakeLee, 2021-11-21:
Can't open esp32s file system...
Device：ESP32S
Mu-editor 1.1.0 beta 6
[image]

log：
Traceback (most recent call last):
  File "C:\Users\Spake\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\base.py", line 683, in ls
    result = tuple(microfs.ls(self.serial))
  File "C:\Users\Spake\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 185, in ls
    out, err = execute(["import os", "print(os.listdir())"], serial)
  File "C:\Users\Spake\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 139, in execute
    raw_on(serial)
  File "C:\Users\Spake\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 91, in raw_on
    flush_to_msg(serial, raw_repl_msg)
  File "C:\Users\Spake\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\contrib\microfs.py", line 72, in flush_to_msg
    raise IOError("Could not enter raw REPL.")
OSError: Could not enter raw REPL.
2021-11-21 13:46:25,855 - mu.interface.main:1038(show_message) DEBUG: There was a problem getting the list of files on the device. Please check Mu's logs for technical information. Alternatively, try unplugging/plugging-in your device and/or restarting Mu.
2021-11-21 13:46:25,855 - mu.interface.main:1039(show_message) DEBUG: None
2021-11-21 13:46:55,955 - mu.logic:1413(show_admin) INFO: Showing admin with logs from C:\Users\Spake\AppData\Local\python\mu\Logs\mu.log
2021-11-21 13:46:55,955 - mu.virtual_environment:917(installed_packages) INFO: Discovering installed third party modules in venv.
2021-11-21 13:46:55,955 - mu.virtual_environment:114(run_blocking) INFO: About to run blocking C:\Users\Spake\AppData\Local\python\mu\mu_venv-38-20211019-201352\scripts\pip.exe with args ['list', '--disable-pip-version-check'] and envvars {}
2021-11-21 13:46:56,387 - mu.virtual_environment:164(wait) DEBUG: Finished: True; exitStatus 0; exitCode 0
2021-11-21 13:46:56,387 - mu.virtual_environment:249(run) DEBUG: Process output: Package           Version
----------------- -------
backcall          0.2.0
bitstring         3.1.9
cffi              1.14.6
click             8.0.1
colorama          0.4.4
cryptography      35.0.0
decorator         5.1.0
ecdsa             0.17.0
entrypoints       0.3
esptool           3.1
Flask             1.1.2
ipykernel         5.5.5
ipython           7.28.0
ipython-genutils  0.2.0
itsdangerous      2.0.1
jedi              0.18.0
Jinja2            3.0.2
jupyter-client    7.0.6
jupyter-core      4.8.1
MarkupSafe        2.0.1
matplotlib-inline 0.1.3
nest-asyncio      1.5.1
numpy             1.21.2
parso             0.8.2
pgzero            1.2.1
pickleshare       0.7.5
pip               20.2.4
prompt-toolkit    3.0.20
pycparser         2.20
pygame            2.0.1
Pygments          2.10.0
pyserial          3.5
python-dateutil   2.8.2
pywin32           301
pyzmq             22.3.0
reedsolo          1.5.4
setuptools        58.2.0
six               1.16.0
tornado           6.1
traitlets         5.1.0
wcwidth           0.2.5
Werkzeug          2.0.1
wheel             0.35.1
2021-11-21 13:46:56,388 - mu.virtual_environment:926(installed_packages) INFO: []

carlosperate, 2021-11-22:
Hi @SpakeLee, thanks for the resport!
Does this happen every time you try to open the files panel?
Does the REPL work?
If you erase the programme in your device does the problem still persist?

SpakeLee, 2021-11-23:
> Hi @carlosperate ，thanks for youyr reply
Does this happen every time you try to open the files panel?
**yes**
Does the REPL work?
**REPK work well**
If you erase the programme in your device does the problem still persist?
**i remove the program in the device then files can work now. thank you.**
[image]

Perritochacarron, 2024-09-06:
hello i have the same problem 
What is the solution?

becky001, 2024-10-03:
I also have that problem

Sunrise17, 2025-08-19:
i have the same problem