mu-editor/mu #1898: Lego MicroPython Failed to Run

stan-ct, 2021-12-01:
### What were you trying to do?

Wrote MicroPython on Lego SPIKE

### What steps did you take to trigger the issue?

Here is the code I wrote:
from spike import PrimeHub
hub = PrimeHub()
hub.left_button.wait_until_pressed()

### What actually happened?

HUB: sync filesystems
HUB: soft reboot
raw REPL; CTRL-B to exit
>OK

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "spike/__init__.py", line 1, in <module>
  File "_api/__init__.py", line 1, in <module>
  File "_api/speaker.py", line 1, in <module>
  File "system/__init__.py", line 1, in <module>
  File "system/callbacks/__init__.py", line 1, in <module>
  File "util/error_handler.py", line 1, in <module>
  File "protocol/__init__.py", line 1, in <module>
  File "protocol/rpc_protocol.py", line 1, in <module>
  File "protocol/notifications.py", line 1, in <module>
RuntimeError: maximum recursion depth exceeded
>
MicroPython v1.14-876-gfbecba865 on 2021-05-04; LEGO Technic Large Hub with STM32F413xx
Type "help()" for more information.
>>> 

### Operating System Version

MacOS 12.0.1

### Mu Version

1.1.0.beta.6

carlosperate, 2021-12-02:
That looks like it's possibly an issue with the lego spike library, have you updated the hub to the latest MicroPython release?

If you go to the REPL and type each line by hand, do you get the same error?

stan-ct, 2021-12-07:
Sorry. There was something wrong with my code. I didn't know the module name was "hub". I considered it was "spike", the same as Lego official App "SPIKE".
Thanks for your help.

carlosperate, 2021-12-07:
Thanks for the update, in that case we can close this issue.
Glad you were able to resolve it!

closed 2021-12-07 by carlosperate

aivarannamaa, 2022-08-20:
This error happens when you import `spike` without importing other dependent modules first. I think LEGO hasn't noticed this (or doesn't care), because they import `hub_runtime` first in the regular setup.

Assuming you don't want to load the rather heavy `hub_runtime`, the best work-around I know, is importing `_api` before importing `spike`.