mu-editor/mu #1901: Crash Report

JuBohm, 2021-12-02:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: en_GB
Platform: Linux 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/mu/app.py", line 159, in run
    raise ex
  File "/usr/local/lib/python3.7/dist-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/usr/local/lib/python3.7/dist-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/usr/local/lib/python3.7/dist-packages/mu/virtual_environment.py", line 768, in create
    self.install_baseline_packages()
  File "/usr/local/lib/python3.7/dist-packages/mu/virtual_environment.py", line 878, in install_baseline_packages
    wheels.download(zipped_wheels_filepath, logger)
  File "/usr/local/lib/python3.7/dist-packages/mu/wheels/__init__.py", line 194, in download
    zip_wheels(zip_filepath, temp_dirpath, logger)
  File "/usr/local/lib/python3.7/dist-packages/mu/wheels/__init__.py", line 165, in zip_wheels
    with zipfile.ZipFile(zip_filepath, "w") as z:
  File "/usr/lib/python3.7/zipfile.py", line 1204, in __init__
    self.fp = io.open(file, filemode)
PermissionError: [Errno 13] Permission denied: '/usr/local/lib/python3.7/dist-packages/mu/wheels/1.1.0.beta.6.zip'

```

JuBohm, 2021-12-02:
[mu.log](https://github.com/mu-editor/mu/files/7643744/mu.log)

JuBohm, 2021-12-02:
Mu don't start now.
What can I do???
Best regards
JuBohm

carlosperate, 2021-12-03:
Hi @JuBohm,
Did you manage to resolve your issue?
Is it possible you installed Mu using `sudo`? In that case you might need to use `sudo` to launch Mu.

JuBohm, 2021-12-03:
Dear Carlos, I have solved the problem, I changed the rights of a subdir
during the installation.
Best regards
JuBohm

2021, 18:08:

carlosperate, 2021-12-03:
That's great to hear, thanks for the update @JuBohm!
I'll close this as resolved then 👍

closed 2021-12-03 by carlosperate