mu-editor/mu #1903: Crash Report

shanimshan, 2021-12-03:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: en_US
Platform: Windows 10 10.0.19042 AMD64

```
, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\DeLL\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\DeLL\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\DeLL\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\DeLL\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 349, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\DeLL\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 840, in setup
    shutil.copytree(path("templates", "web/"), template_path)
  File "C:\Users\DeLL\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 557, in copytree
    return _copytree(entries=entries, src=src, dst=dst, symlinks=symlinks,
  File "C:\Users\DeLL\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 458, in _copytree
    os.makedirs(dst, exist_ok=dirs_exist_ok)
  File "C:\Users\DeLL\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\DeLL\\mu_code\\templates'

```

carlosperate, 2021-12-04:
Thanks for the report @shanimshan!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\DeLL\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-12-17:
Hi @shanimshan, I suspect this problem might have been caused by #1755.

If you are still experiencing this error, are you able to find the logs and send them to us? File should be located in: `C:\Users\DeLL\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2022-03-07:
Hi @shanimshan, are you still encountering this error?

closed 2022-05-31 by carlosperate