mu-editor/mu #1904: Crash Report

bluehobo, 2021-12-04:
Win 10, company managed (ive got admin rights thought). intel architecture

**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: nb_NO
Platform: Windows 10 10.0.19042 AMD64

```
03, in ensure_and_create
    self.create()
  File "C:\Users\sir\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 768, in create
    self.install_baseline_packages()
  File "C:\Users\sir\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 880, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "C:\Users\sir\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Users\sir\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Users\sir\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Users\sir\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\sir\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 177, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process did not terminate normally:

```

bluehobo, 2021-12-04:
Happens on first run after fresh install. Might be permission related?

bluehobo, 2021-12-04:
[mu.log](https://github.com/mu-editor/mu/files/7654023/mu.log)

carlosperate, 2021-12-17:
Hi @bluehobo,

Some people have reported a similar issue caused by a running antivirus. Do you have an antivirus installed, and if you do, would you be able to disable it and try again?

If that works it'd be good to know what antivirus software it is so that we can try to replicate and see if we can resolve the issue without having to disable it.

bluehobo, 2021-12-17:
No 3rd party antivirus, but I'm on a company managed computer. I do have local administrator rights though

carlosperate, 2021-12-17:
Thanks for the quick update @bluehobo.

I think it's very unlikely that this is caused by Windows Defender, as otherwise we'd see a lot more reports with this issue, but because we don't know yet what antivirus software the other users disabled it'd be good to test it as well.

Could you try disabling the real time protection of Windows Defender just in case that makes a difference?

After that if you could copy the latest version of the log that would be great as well. We might want to run in a terminal the last command that failed, which will likely be something like `C:\Users\sir\AppData\Local\python\mu\mu_venv-38-SOME_DATE_HERE-SOME_OTHER_NUMBER_HERE\scripts\pip.exe install --disable-pip-version-check --no-deps --no-index C:\Users\sir\AppData\Local\Temp\SOME_TEMP_FOLDER\backcall-0.2.0-py2.py3-none-any.whl`

bluehobo, 2021-12-17:
I'm sorry, I've reversed to a stable release now and out of office until Christmas. I won't be able to do that test anytime soon.

carlosperate, 2021-12-17:
No worries, if you can run it when you are back that would still be useful.
Btw, what version did you install that worked?

bluehobo, 2021-12-17:
I think it was 1.0.4 but have to double check. It is the last of the stable ones, in with .exe installation. I don't think I can disable anything on the ms security centre on my company machine, I think access to those are disabled even if I have local privileges. I also have a hunch that this is the cause of the problem. Instead I will try to install the latest beta on my private machine and see if that works!

bluehobo, 2021-12-22:
Sorry for late reply. I just installed latest version on my personal device (Surface Pro 4) and no problems whatsoever.

Again, I suspect the problems on my work computer is because of insufficient local permissons on my company managed computer.

carlosperate, 2021-12-22:
No worries, we really appreciate the testing!
It's definitely good to know it works on your personal device,. After the break, if you still can, it would be interesting to continue investigating what might be the issue. Ideally it's something we can fix, or at least workaround so Mu doesn't crash. We are invested in making sure Mu works in computers with local restrictions, as that is quite common in education environments.

bluehobo, 2022-03-15:
I just tried version 1.1.1 on my enterprise win 10 computer and get the same error.

carlosperate, 2022-03-16:
Thanks @bluehobo, are you able to attach the Mu log here? It should be in `%LocalAppData%\python\mu\Logs\mu.log`

bluehobo, 2022-03-16:
I'm sorry, I forgot about that, was stressing with a project for my class, so I reverted to version 1.1.1 which works fine. However, it was the same error message as in the report at the top of this thread. So, the problem might be related to the msi-installer. It does not ask me for admin privileges when installing, but the .exe installer does and that works fine. Hope this is helpful!