mu-editor/mu #1906: Crash Report

yasenstar, 2021-12-05:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: en_US
Platform: Linux 5.4.139-16311-g330b3df0010b #1 SMP PREEMPT Tue Oct 26 20:44:34 PDT 2021 x86_64

```
Traceback (most recent call last):
  File "/home/shadoedragon/.local/lib/python3.7/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/shadoedragon/.local/lib/python3.7/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/shadoedragon/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 621, in ensure_and_create
    self.recreate()
  File "/home/shadoedragon/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 566, in recreate
    self.create()
  File "/home/shadoedragon/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 768, in create
    self.install_baseline_packages()
  File "/home/shadoedragon/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 878, in install_baseline_packages
    wheels.download(zipped_wheels_filepath, logger)
  File "/home/shadoedragon/.local/lib/python3.7/site-packages/mu/wheels/__init__.py", line 192, in download
    pip_download(temp_dirpath, logger)
  File "/home/shadoedragon/.local/lib/python3.7/site-packages/mu/wheels/__init__.py", line 130, in pip_download
    "Pip was unable to download %s" % pip_identifier
mu.wheels.WheelsDownloadError: Pip was unable to download esptool==3.*

```
[mu.log](https://github.com/mu-editor/mu/files/7656298/mu.log)

carlosperate, 2021-12-06:
Thanks for the report and attaching the log @yasenstar!

Okay, so the interesting part of the log is that is trying to build `cffi` from source:

```
  Building wheels for collected packages: cffi
    Running setup.py bdist_wheel for cffi: started
    Running setup.py bdist_wheel for cffi: finished with status 'error'
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-q6utnnn2/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-jmd5zjp8 --python-tag cp37:
    running bdist_wheel
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-3.7
    creating build/lib.linux-x86_64-3.7/cffi
    copying cffi/__init__.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/api.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/backend_ctypes.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/cffi_opcode.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/commontypes.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/cparser.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/error.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/ffiplatform.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/lock.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/model.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/pkgconfig.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/recompiler.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/setuptools_ext.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/vengine_cpy.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/vengine_gen.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/verifier.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/_cffi_include.h -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/parse_c_type.h -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/_embedding.h -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/_cffi_errors.h -> build/lib.linux-x86_64-3.7/cffi
    running build_ext
    building '_cffi_backend' extension
    creating build/temp.linux-x86_64-3.7
    creating build/temp.linux-x86_64-3.7/c
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DUSE__THREAD -DHAVE_SYNC_SYNCHRONIZE -I/usr/include/ffi -I/usr/include/libffi -I/usr/include/python3.7m -c c/_cffi_backend.c -o build/temp.linux-x86_64-3.7/c/_cffi_backend.o
    c/_cffi_backend.c:15:10: fatal error: ffi.h: No such file or directory
     #include <ffi.h>
              ^~~~~~~
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    ----------------------------------------
    Failed building wheel for cffi
    Running setup.py clean for cffi
  Failed to build cffi
  Installing collected packages: setuptools, wheel, pycparser, cffi, semantic-version, typing-extensions, tomli, setuptools-rust
    Running setup.py install for cffi: started
      Running setup.py install for cffi: finished with status 'error'
      Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-q6utnnn2/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-2jt2h6ug/install-record.txt --single-version-externally-managed --prefix /tmp/pip-build-env-2nm983lu --compile:
      running install
      running build
      running build_py
      creating build
      creating build/lib.linux-x86_64-3.7
      creating build/lib.linux-x86_64-3.7/cffi
      copying cffi/__init__.py -> build/lib.linux-x86_64-3.7/cffi
      copying cffi/api.py -> build/lib.linux-x86_64-3.7/cffi
      copying cffi/backend_ctypes.py -> build/lib.linux-x86_64-3.7/cffi
      copying cffi/cffi_opcode.py -> build/lib.linux-x86_64-3.7/cffi
      copying cffi/commontypes.py -> build/lib.linux-x86_64-3.7/cffi
      copying cffi/cparser.py -> build/lib.linux-x86_64-3.7/cffi
      copying cffi/error.py -> build/lib.linux-x86_64-3.7/cffi
      copying cffi/ffiplatform.py -> build/lib.linux-x86_64-3.7/cffi
      copying cffi/lock.py -> build/lib.linux-x86_64-3.7/cffi
      copying cffi/model.py -> build/lib.linux-x86_64-3.7/cffi
      copying cffi/pkgconfig.py -> build/lib.linux-x86_64-3.7/cffi
[... 22 lines cut ...]
Command "/usr/bin/python3 -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-2nm983lu --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --find-links /tmp/tmp_1x9opeh -- setuptools>=40.6.0 wheel "cffi>=1.12; platform_python_implementation != 'PyPy'" setuptools-rust>=0.11.4" failed with error code 1 in None
2021-12-05 10:55:33,738 - root:171(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/home/shadoedragon/.local/lib/python3.7/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/shadoedragon/.local/lib/python3.7/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/shadoedragon/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 621, in ensure_and_create
    self.recreate()
[... cut ...]
  File "/home/shadoedragon/.local/lib/python3.7/site-packages/mu/wheels/__init__.py", line 130, in pip_download
    "Pip was unable to download %s" % pip_identifier
mu.wheels.WheelsDownloadError: Pip was unable to download esptool==3.*
```
 
pip didn't find a wheel for your platform, which is a bit odd as v1.15.0 has `cffi-1.15.0-cp37-cp37m-manylinux_2_12_x86_64.manylinux2010_x86_64.whl`.

@yasenstar did you install Mu in a virtual environment or on the system Python? what version of pip are you running? could you update pip and try again?

carlosperate, 2021-12-16:
Hi @yasenstar, just wondering if you had a chance to update pip and try again?