mu-editor/mu #1909: Crash Report

chee57, 2021-12-07:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: en_US
Platform: Windows 10 10.0.18363 AMD64

```
le "C:\Users\Carol\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 339, in installed
    next(iterlines)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Carol\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\Carol\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Carol\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Users\Carol\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 769, in create
    self.register_baseline_packages()
  File "C:\Users\Carol\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 887, in register_baseline_packages
    packages = list(self.pip.installed())
  File "C:\Users\Carol\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 345, in installed
    raise VirtualEnvironmentError("Unable to parse installed packages")
mu.virtual_environment.VirtualEnvironmentError: Unable to parse installed packages

```

chee57, 2021-12-07:
This happens when I try to start the program.  Please help!

[mu.log.txt](https://github.com/mu-editor/mu/files/7671882/mu.log.txt)

carlosperate, 2021-12-08:
Thank you for the report and the log @chee57!

Looking at the log this is the important part, it installs all the packages in the virtual environment, but then it fails on the `pip list` step:
```
2021-12-07 14:42:45,678 - mu.virtual_environment:850(install_from_zipped_wheels) INFO: About to install from wheel: Werkzeug-2.0.1-py3-none-any.whl
2021-12-07 14:42:45,678 - mu.virtual_environment:114(run_blocking) INFO: About to run blocking C:\Users\Carol\AppData\Local\python\mu\mu_venv-38-20211207-143653\scripts\pip.exe with args ['install', '--disable-pip-version-check', '--no-deps', '--no-index', 'C:\\Users\\Carol\\AppData\\Local\\Temp\\tmp467smqx2\\Werkzeug-2.0.1-py3-none-any.whl'] and envvars {}
2021-12-07 14:43:00,858 - mu.virtual_environment:164(wait) DEBUG: Finished: True; exitStatus 0; exitCode 0
2021-12-07 14:43:00,858 - mu.virtual_environment:249(run) DEBUG: Process output: 
2021-12-07 14:43:00,999 - mu.virtual_environment:114(run_blocking) INFO: About to run blocking C:\Users\Carol\AppData\Local\python\mu\mu_venv-38-20211207-143653\scripts\pip.exe with args ['list', '--disable-pip-version-check'] and envvars {}
2021-12-07 14:43:03,233 - mu.virtual_environment:164(wait) DEBUG: Finished: True; exitStatus 0; exitCode 0
2021-12-07 14:43:03,233 - mu.virtual_environment:249(run) DEBUG: Process output: 
2021-12-07 14:43:03,233 - mu.virtual_environment:643(ensure_and_create) ERROR: Unable to parse installed packages
2021-12-07 14:43:03,702 - mu.virtual_environment:540(quarantine_venv) INFO: Quarantined C:\Users\Carol\AppData\Local\python\mu\mu_venv-38-20211207-143653 as C:\Users\Carol\AppData\Local\python\mu\mu_venv-38-20211207-143653.FAILED
2021-12-07 14:43:10,718 - root:171(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\Carol\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 339, in installed
    next(iterlines)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Carol\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\Carol\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Carol\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Users\Carol\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 769, in create
    self.register_baseline_packages()
  File "C:\Users\Carol\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 887, in register_baseline_packages
    packages = list(self.pip.installed())
  File "C:\Users\Carol\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 345, in installed
    raise VirtualEnvironmentError("Unable to parse installed packages")
mu.virtual_environment.VirtualEnvironmentError: Unable to parse installed packages
``` 

@chee57 could you run this in a command prompt and paste the result in this GitHub issue? It'd be really useful to see the pip output:

```
C:\Users\Carol\AppData\Local\python\mu\mu_venv-38-20211207-143653\scripts\pip.exe list --disable-pip-version-check
```

carlosperate, 2021-12-15:
@chee57 It might be possible that an antivirus is interfering with some Mu operations. Do you have an antivirus running? If you do, would be able to deactivate it when launching Mu to see if the problem persists?

chee57, 2021-12-15:
Thank you for your response.  It was the antivirus.

carlosperate, 2021-12-16:
Thanks for the update @chee57, what Antivirus software do you use? It'll be useful to know to try to replicate the problem to see if we can solve it without deactivating it.