mu-editor/mu #1912: Mixed line endings (EOL) when editing on Windows

devdanzin, 2021-12-08:
### What were you trying to do?

Input text into Mu and get consistent EOL

### What steps did you take to trigger the issue?

I used `EditorPane.setEolVisibility(True)` so Mu displays EOL when editing.

### What did you expect to happen?

Mu should use `LF` as EOL

### What actually happened?

On Windows, despite Mu working with internal `LF` EOL, editing text actually adds `CRLF` line endings. It mostly hasn't been an issue, but was the cause of #1511 and goes against expectations, since we should be working with Mu's internal EOL and code accordingly.

### Operating System Version

Windows 10

### Mu Version

1.1.0.beta.6

### Other Info

I propose that we set `EditorPane.setEolMode(QsciScintilla.EolUnix)` so we're always working with `LF`, even on Windows. It shouldn't affect which EOL the file will be saved with, as that is determined on load or creation.

carlosperate, 2021-12-09:
> It shouldn't affect which EOL the file will be saved with, as that is determined on load or creation.

Yeah, I think this is the important part, as long as the file is saved with the platform EOL it should be fine.

Does `EditorPane.setEolMode(QsciScintilla.EolUnix)` convert the EOLs when text pasted into the editor?

devdanzin, 2021-12-09:
> Does `EditorPane.setEolMode(QsciScintilla.EolUnix)` convert the EOLs when text pasted into the editor?

Yes it does. And currently pasted EOLs  are converted to `CRLF`, as we run with the equivalent to `EditorPane.setEolMode(QsciScintilla.EolWindows)`.

As a side note, besides saving and reopening the file, using the "Tidy" action will also convert all EOLs to `LF`, so a workaround for #1511 would be tidying the code before toggling comment.

carlosperate, 2021-12-10:
> > Does `EditorPane.setEolMode(QsciScintilla.EolUnix)` convert the EOLs when text pasted into the editor?
> 
> Yes it does. And currently pasted EOLs are converted to `CRLF`, as we run with the equivalent to `EditorPane.setEolMode(QsciScintilla.EolWindows)`.

Great!

> As a side note, besides saving and reopening the file, using the "Tidy" action will also convert all EOLs to `LF`, so a workaround for #1511 would be tidying the code before toggling comment.

The tidy action could significantly reorganise the user script, which might be quite unexpected when trying to only comment/uncomment a line or block.