mu-editor/mu #1914: Crash Report

Botlol1, 2021-12-09:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: en_GB
Platform: Windows 10 10.0.19044 AMD64

```

    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "C:\Users\Gamer 1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Users\Gamer 1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Users\Gamer 1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Users\Gamer 1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\Gamer 1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 101:
Unable to create process using 'C:\Users\Gamer 1\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\Gamer 1\AppData\Local\python\mu\mu_venv-38-20211209-201349\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\GAMER1~1\AppData\Local\Temp\tmpoadvgbwf\backcall-0.2.0-py2.py3-none-any.whl'

```

tjguk, 2021-12-13:
This looks related to #1920 where we've also failed to install `backcall-0.2.0`. Not sure what's using that as it's not one of our direct deps

carlosperate, 2021-12-15:
Thanks for the report @Botlol1, would you be able to attach the Mu log to this GitHub issue? It'd be really useful to us, it should be in this path: C:\Users\Gamer 1\AppData\Local\python\mu\Logs\mu.log