mu-editor/mu #1917: Crash Report

wbrzezki, 2021-12-10:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: pl_PL
Platform: Windows 10 10.0.22000 AMD64

```
le "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 339, in installed
    next(iterlines)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
[mu.log](https://github.com/mu-editor/mu/files/7694671/mu.log)

  File "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 769, in create
    self.register_baseline_packages()
  File "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 887, in register_baseline_packages
    packages = list(self.pip.installed())
  File "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 345, in installed
    raise VirtualEnvironmentError("Unable to parse installed packages")
mu.virtual_environment.VirtualEnvironmentError: Unable to parse installed packages
[mu.log](https://github.com/mu-editor/mu/files/7694655/mu.log)

```

tjguk, 2021-12-13:
Thanks for reporting this, @wbrzezki -- are you able to upload the full logs? They'll be in `%LOCALAPPDATA%\python\mu\logs`

tjguk, 2021-12-13:
Oddly, it looks as though you tried to upload them -- I can see a marker in that first report. But I can't see the logs in the web interface.

wbrzezki, 2021-12-13:
Thank you for your answer @tjguk. Fortunately, I found the cause of the issue. My antivirus was blocking Mu editor from starting.
I think I can close this issue.

carlosperate, 2021-12-16:
Thanks for the update @wbrzezki.

What antivirus do you run?