mu-editor/mu #1919: Crash Report

chaliudo, 2021-12-11:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: en_US
Platform: Linux 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l

```
Traceback (most recent call last):
  File "/home/pi/mu-source/mu/modes/python3.py", line 109, in start_kernel
    self.repl_kernel_manager.start_kernel()
  File "/home/pi/mu-source/mu/modes/python3.py", line 42, in start_kernel
    kernel_cmd, kw = self.pre_start_kernel(**kw)
AttributeError: 'MuKernelManager' object has no attribute 'pre_start_kernel'

```

tjguk, 2021-12-13:
Thanks for reporting, @chaliudo -- this one's surprising. To help us out, do you have the logs, please? 

It looks like you're on a Raspberry Pi; is that right? We could do with knowing what version of the qtconsole package you've got installed within the environment from which Mu is running. (It should be 4.7.7 but it would be good to know if that's what you've got..)

carlosperate, 2021-12-17:
I think this is the issue discussed in PR https://github.com/mu-editor/mu/pull/1517, so we are still figuring out the best solution.

In the meantime, the easiest way to fix this would be to update the `jupyter_client` package to a newer version.
However, since you might have an old version because some other application in your OS needs it, there is a risk that updating it could cause issues in that other application. Updating your system `jupyter_client` is the easiest solution, but it does have that risk.

You can install Mu in a virtual environment, this isolates the Mu installation, so that its dependencies are not combined with the system packages. 
If you are familiar with virtual environments you can create a new one for Mu, install Mu using pip and it should pick up the latest version of `jupyter_client`.

Alternatively I like to use the pipx tool, which simplifies all this process: https://pipxproject.github.io/pipx/
If you install it you can then use pipx to automatically install Python applications (like Mu) in their own isolated virtual environment, and they can be run without manually activating the environment:
- Install pipx
    - Update pip: `python3 -m pip install --upgrade pip` 
    - Run: `python3 -m pip install --user pipx`
    - Run: `python3 -m pipx ensurepath`
- After installation you might need to close all your terminals and open a new one (or maybe even reboot, the last command should print out information)
- Now that pipx is installed and configured you can use it to install Mu: `pipx install mu-editor==1.1.0b5`
- And you can run mu: `mu-editor`