mu-editor/mu #1922: " compatible" string is no good for translation

rffontenelle, 2021-12-13:
### What were you trying to do?

Translate Mu

### What steps did you take to trigger the issue?

I bootstrapped translation by running `make translate_begin LANG=pt_BR` and then started translating with my preferred PO editor.

After some strings, I hit `" compatible"` string. Its source location is in

https://github.com/mu-editor/mu/blob/12d80470add3f790a0a38cc559106a618773f3b3/mu/logic.py#L576

It is used in `def name`, so I believe it returns "SOME_BOARD compatible".

### What did you expect to happen?

Be able to translate the

### What actually happened?

The string `" compatible"` is impossible to be translated. In Portuguese, when something is FOOBAR-compatible, I have to translate reordering to something like "compatible with FOOBAR" (but in Portuguese, obviously). With the current partial string I can't do this correctly,

### Operating System Version

Ubuntu 20.04 LTS WSL2

### Mu Version

N/A

### Other Info

I suggest to have it rephrased with placeholders. E.g. `" {boardname} compatible"`

carlosperate, 2021-12-17:
Thanks @rffontenelle!

I think we already use `{}` to format strings, so after merging the two Portuguese translations we can create a PR updating the English string and Portuguese translations.

carlosperate, 2021-12-20:
Opened PR https://github.com/mu-editor/mu/pull/1936.

commit 005a15f8 (2021-12-20): Fixes #1922 - fix string for better translatability.
commit 9a7cd602 (2021-12-20): Merge pull request #1936 from mu-editor/str-fix

Fixes #1922 - fix string for better translatability.
closed 2021-12-20 by ntoll