mu-editor/mu #1925: Správa o zlyhaní

RaRo100, 2021-12-14:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: sk_SK
Platform: Windows 10 10.0.18362 AMD64

```
Traceback (most recent call last):
  File "C:\Users\bosorka\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\bosorka\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\bosorka\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\bosorka\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\bosorka\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

```

carlosperate, 2021-12-15:
Hi @RaRo100, thanks for the report!
Could you update to v1.1.0.beta.6? Is the latest release and likely resolves you issue: https://github.com/mu-editor/mu/releases

carlosperate, 2022-03-07:
Hi @RaRo100, could you update to the latest release and see if that version has resolved your issue? https://github.com/mu-editor/mu/releases/tag/v1.1.1
Thanks!

carlosperate, 2022-05-31:
Closing due to lack of activity.

closed 2022-05-31 by carlosperate