mu-editor/mu #1928: Crash Report

mpgprofe, 2021-12-16:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: es_ES
Platform: Windows 10 10.0.19044 AMD64

```
Traceback (most recent call last):
  File "C:\Users\manue\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\manue\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\manue\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Users\manue\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 767, in create
    self.create_venv()
  File "C:\Users\manue\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 798, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\manue\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\manue\AppData\Local\python\mu\mu_venv-38-20211216-182936
ModuleNotFoundError: No module named 'virtualenv.activation.nushell'

```

carlosperate, 2021-12-16:
Thanks for the report @mpgprofe!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\manue\AppData\Local\python\mu\Logs\mu.log`

I think that this issue might be the same as https://github.com/mu-editor/mu/issues/1831, we are trying one possible solution. Would you be able to try the installer at the bottom of this page, in the "Artifacts" section to see if it resolves it for you?
https://github.com/mu-editor/mu/actions/runs/1574840282

To ensure things go smoothly it'd be highly recommended to first uninstall the version of Mu in your PC before installing this version.

If you could try this installer it would be very useful to us to confirm we have a fix for the issue.

carlosperate, 2022-05-31:
Closing due to lack of activity.