mu-editor/mu #1933: Crash Report

james-bryant, 2021-12-17:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: en_US
Platform: Windows 10 10.0.22000 AMD64

```
Traceback (most recent call last):
  File "C:\Users\James\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\James\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\James\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Users\James\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 767, in create
    self.create_venv()
  File "C:\Users\James\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 798, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\James\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\James\AppData\Local\python\mu\mu_venv-38-20211217-101920
ModuleNotFoundError: No module named 'virtualenv.activation.nushell'

```

carlosperate, 2021-12-17:
Thanks for the report @james-bryant, I think this crash might be due to this issue: https://github.com/mu-editor/mu/issues/1831

We think we might have found a solution to this problem. Would you be able to try the installer at the bottom of this page, in the "Artifacts" section to see if it resolves the problem you are encountering?
https://github.com/mu-editor/mu/actions/runs/1574840282

To ensure things go smoothly it'd be highly recommended to first uninstall the version of Mu in your PC before installing this version.

If you could try this installer it would be very very useful to us to confirm the issue has been fixed.

james-bryant, 2021-12-20:
That does appear to have resolved the issue.

Thanks,
James

carlosperate, 2021-12-20:
That's great, thanks for testing @james-bryant! 

As the fix will be present in the upcoming next release, and the problem is tracked in https://github.com/mu-editor/mu/issues/1831, I'll close this crash report.

Thanks for the report and testing, it has been really useful!

closed 2021-12-20 by carlosperate