mu-editor/mu #1934: Installing CircuitPython or MicroPython on a new ESP-C3-32S

profharris, 2021-12-17:
### What were you trying to do?

Trying to flash CircuitPython or MicroPython on a new ESP-C3-32S

### What steps did you take to trigger the issue?

I updated to Mu 1.1.0.beta.6.  I downloaded the latest .bin files for both CircuitPython or MicroPython. Connecting to the ESP-C3-32S, and erasing the flash goes without any problems. Then for both .bin files Mu Administration responds with...

"C:\Users\mphar\AppData\Local\python\mu\mu_venv-38-20211217-125954\scripts\python.exe" "-mesptool" --port COM2 erase_flash
esptool.py v3.1
Serial port COM2
Connecting....
Detecting chip type... ESP32-C3
Chip is unknown ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 84:f7:03:a5:43:e0
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 17.5s
Hard resetting via RTS pin...

"C:\Users\mphar\AppData\Local\python\mu\mu_venv-38-20211217-125954\scripts\python.exe" "-mesptool" --chip esp32 --port COM2 --baud 115200 write_flash -z 0x1000 "C:\Users\mphar\OneDrive - Del Mar College\ESP32_ESP8266\NodeMCU ESP-C3-32S\zips\adafruit-circuitpython-ai_thinker_esp32-c3s-en_US-7.1.0-beta.3.bin"
esptool.py v3.1
Serial port COM2
Connecting....

A fatal error occurred: This chip is ESP32-C3 not ESP32. Wrong --chip argument?

Error on flashing. Aborting.

### What did you expect to happen?

To flash the CircuitPython or MicroPython like I've done for SAMD-21s, ESP32-WROOM-32s, ESP-WROOM-32s, and ESP32Ds.

### What actually happened?

upgraded to Mu 1.1.0.beta.6    (from .4)
Trying to flash/upload CircutPython...

Mu Administration:

"C:\Users\mphar\AppData\Local\python\mu\mu_venv-38-20211217-125954\scripts\python.exe" "-mesptool" --port COM2 erase_flash
esptool.py v3.1
Serial port COM2
Connecting....
Detecting chip type... ESP32-C3
Chip is unknown ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 84:f7:03:a5:43:e0
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 17.5s
Hard resetting via RTS pin...

"C:\Users\mphar\AppData\Local\python\mu\mu_venv-38-20211217-125954\scripts\python.exe" "-mesptool" --chip esp32 --port COM2 --baud 115200 write_flash -z 0x1000 "C:\Users\mphar\OneDrive - Del Mar College\ESP32_ESP8266\NodeMCU ESP-C3-32S\zips\adafruit-circuitpython-ai_thinker_esp32-c3s-en_US-7.1.0-beta.3.bin"
esptool.py v3.1
Serial port COM2
Connecting....

A fatal error occurred: This chip is ESP32-C3 not ESP32. Wrong --chip argument?

Error on flashing. Aborting.

--------------------------------------------------------------------------------
[... 202 lines cut ...]
esptool.py v3.1
Serial port COM2
Connecting....
Detecting chip type... ESP32-C3
Chip is unknown ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 84:f7:03:a5:43:e0
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 17.1s
Hard resetting via RTS pin...

"C:\Users\mphar\AppData\Local\python\mu\mu_venv-38-20211217-125954\scripts\python.exe" "-mesptool" --chip esp32 --port COM2 --baud 115200 write_flash -z 0x1000 "C:\Users\mphar\OneDrive - Del Mar College\ESP32_ESP8266\NodeMCU ESP-C3-32S\zips\esp32c3-20210902-v1.17.bin"
esptool.py v3.1
Serial port COM2
Connecting....

A fatal error occurred: This chip is ESP32-C3 not ESP32. Wrong --chip argument?

Error on flashing. Aborting.

-profharris

carlosperate, 2021-12-17:
Thanks for the report @profharris.
Assuming that ESP32-C3 might require new flags then support hasn't been added to Mu yet.

I don't think anybody from the maintainers teams has one of these devices, would you be happy to work on a PR to add support?
Is this the RISC-V version with integrated USB? In the pictures it looks like the board might still have a USB-to-UART bridge. Either way I assume the `esptool` should work the same in either mode? Or does it need more flags to be configured depending on the board?

The Mu settings dialog could be update here to add another entry to the "Choose device type" menu, and that would configure which flags are sent to `esptool`:
https://github.com/mu-editor/mu/blob/20186eeecc18879a1e17c828f6ce553376c25301/mu/interface/dialogs.py#L282

And if C3 support was added in a specific version of `esptool` then we should add that as well to this line: https://github.com/mu-editor/mu/blob/20186eeecc18879a1e17c828f6ce553376c25301/mu/wheels/__init__.py#L48

profharris, 2021-12-19:
I'm on vacation to see my daughter. I'll test the enhancement-mode: esp modifications to Mu when I return on Thursday.  Thank-you for your prompt attention to my report.
-profharris

carlosperate, 2021-12-19:
No rush, it'll be welcome whenever you have the time, thank you! And have a nice vacation!

thebigboss84, 2022-01-30:
you have to change --chip argument from esp32 to esp32-c3 .

andypiper, 2022-02-23:
The correct format is

`esptool.py --chip esp32c3 [other stuff]`

I'm doing a lot with the C3 boards at the moment, it would be great to have this fixed...

andypiper, 2022-03-19:
Also noting that the [new Wemos C3 Mini boards](https://www.wemos.cc/en/latest/c3/c3_mini.html) ship with MicroPython so it would be good to see this updated in Mu :-)