mu-editor/mu #1935: can't compile mu on macOS 12.1 with M1 chip

algon-ns, 2021-12-18:
### What were you trying to do?

install mu

### What steps did you take to trigger the issue?

normal procedure, which failed,
the I went to ~/.emacs.d/straight/repso/mu and did make......

### What did you expect to happen?

compiling worked

### What actually happened?

failure Can't find xapian.h for several modules

xapian is installed and is in /opt/homebrew/   and Xapian.h in ./include

### Operating System Version

macOS 12.1 (with M1 chip)

### Mu Version

most recent (pulled in from GitHub)

### Other Info

[image]

### Editor Log

can't find any log files on mu

carlosperate, 2021-12-18:
Hi @algon-ns, thanks for the report!

That looks very wrong, are you sure you cloned the right repository?
The default target in the Makefile should have simply printed all the options:

```
 $ make

There is no default Makefile target right now. Try:

make run - run the local development version of Mu.
make clean - reset the project and remove auto-generated assets.
make flake8 - run the flake8 code checker.
make test - run the test suite.
make coverage - view a report on test coverage.
make tidy - tidy code with the 'black' formatter.
make check - run all the checkers and tests.
make dist - make a dist/wheel for the project.
make publish-test - publish the project to PyPI test instance.
make publish-live - publish the project to PyPI production.
make docs - run sphinx to create project documentation.
make translate_begin LANG=xx_XX - create/update a mu.po file for translation.
make translate_done LANG=xx_XX - compile translation strings in mu.po to mu.mo file.
make translate_test LANG=xx_XX - run translate_done and launch Mu in the given LANG.
make win32 - create a 32bit Windows installer for Mu.
make win64 - create a 64bit Windows installer for Mu.
make macos - create a macOS native application for Mu.
make video - create an mp4 video representing code commits.
```

The screenshot you posted is invoking Ninja with a C++ application, which I don't think it's related to the Mu Editor at all.

As a Python application you don't have to compile it to use it, you can find more developer info in the docs: https://mu.readthedocs.io/en/latest/

algon-ns, 2021-12-18:
Hi,

The file mu4e.info <http://mu4e.info/> file starts like this_

Dette er
/Users/niels/nec.emacs.d/straight/repos/mu/build/mu4e/mu4e.info,
produceret med makeinfo version 4.8 ud fra
/Users/niels/nec.emacs.d/straight/repos/mu/mu4e/mu4e.texi.

   Copyright (C) 2012-2020 Dirk-Jan C. Binnema

     Permission is granted to copy, distribute and/or modify this
     document under the terms of the GNU Fre……

And the dried looks like this…

  /Users/niels/.emacs.d/straight/repos/mu/build:
  [ Directories ]
  drwxr-xr-x  5 niels  160 2021-11-10 16:18 guile
  drwxr-xr-x 16 niels  512 2021-10-31 14:57 lib
  drwxr-xr-x  2 niels   64 2021-10-31 14:57 man
  drwxr-xr-x 12 niels  384 2021-12-11 18:29 meson-info
  -rw-r--r--  1 niels  14K 2021-12-11 18:29 meson-logs/meson-log.txt
  drwxr-xr-x 15 niels  480 2021-12-11 18:29 meson-private
  drwxr-xr-x  6 niels  192 2021-10-31 14:57 mu
  drwxr-xr-x  4 niels  128 2021-12-11 18:29 mu4e
  [ Default ]
  -rw-r--r--  1 niels   92 2021-10-31 14:57 .gitignore
  -rw-r--r--  1 niels  108 2021-10-31 14:57 .hgignore
  -rw-r--r--  1 niels  50K 2021-12-18 15:26 .ninja_deps
  -rw-r--r--  1 niels 2,0K 2021-12-18 15:26 .ninja_log
  -rw-r--r--  1 niels  91K 2021-12-11 18:29 build.ninja
  -rw-r--r--  1 niels  59K 2021-12-11 18:29 compile_commands.json
  -rw-r--r--  1 niels  317 2021-11-10 16:25 config.h
  -rw-r--r--  1 niels  101 2021-12-11 18:18 version.texi

Kind regards
Niels. Who is confused on a very high level….

 <applewebdata://107FDC7D-4377-48A7-962E-05F14F3CE36F>
Niels Søndergaard
Mariehøj 236, 2990 Nivå
 <tel:+45%204052%202789>+45 4052 2789 ***@***.*** ***@***.***>
Ultra posse nemo obligatur.

carlosperate, 2021-12-18:
Hi @algon-ns, if you are trying to compile mu4e (looks like an emacs email client?) you are in the wrong repository. This is for the Mu Editor: https://codewith.mu

algon-ns, 2021-12-19:
Pls see under the closed issue 1935 with info

Kind regards
Niels

 <applewebdata://DE50CAC3-5C93-4B22-BEDE-4BFA6F2F9FA6>
Niels Søndergaard
Mariehøj 236, 2990 Nivå
 <tel:+45%204052%202789>+45 4052 2789 ***@***.*** ***@***.***>
Ultra posse nemo obligatur.

closed 2021-12-19 by carlosperate