mu-editor/mu #1940: Crash Report

johnsokol, 2021-12-22:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: en_US
Platform: Windows 10 10.0.19043 AMD64

```
Traceback (most recent call last):
  File "C:\Users\johns\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\johns\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\johns\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\johns\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\johns\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 349, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\johns\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 809, in setup
    os.makedirs(wd)
  File "C:\Users\johns\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\johns\\mu_code'

```

carlosperate, 2021-12-23:
Thanks for the report @johnsokol!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\johns\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2022-05-31:
Hi @johnsokol,

I believe you've encountered this issue https://github.com/mu-editor/mu/issues/1755

Could you try opening Mu again, but ensuring it is only executed once?
I believe the problem arises when the Mu application is opened more than once when running for the first time. It might take a some time for the first launch to complete, but after that it should be much quicker.

Before doing that, if you could attach to this GitHub issue the original log file that would be really useful as well. It should be located in `C:\Users\johns\AppData\Local\python\mu\Logs\mu.log`.